YosysHQ/nextpnr #896: Could not initialize GLX

ChSotiriou, 2022-01-19:
Just installed the OSS CAD Suite and nextpnr can't open the GUI. It exists with the following error.

[image]

Please let me know if further information is needed.

ChSotiriou, 2022-01-20:
Issue resolved by removing the OSS CAD Suite and compiling from source.

closed 2022-01-20 by ChSotiriou

mmicko, 2022-01-20:
Sorry for late reaction. If you could please provide which distribution are you using, in order to  try to replicate this and solve this for future. Thanks.

ChSotiriou, 2022-01-20:
I'm using Arco Linux. The version of nextpnr that came with oss-cad-suite (installed from the AUR) is:

```
➜  /opt/oss-cad-suite/bin🔒 ./nextpnr-ice40 --version
nextpnr-ice40 -- Next Generation Place and Route (Version fd2d4a8f)
```