YosysHQ/nextpnr #910: Ice40 UP5K: embedded block RAM clock edge/polarity

jswrightoc, 2022-02-06:
It looks like the NegClk configuration bits on the UP5K for the read and write ports may be swapped.

I can verify the read clock polarity is not affected when a SB_RAM40_4KNR is instantiated, but is altered when a SB_RAM40_4KNW cell is used instead. The timing analysis will be incorrect as well. I have not run any tests to be able to infer the clock polarity of the write port.

I get the expected clock behavior testing with the HX1K, although I stumbled across a different potential issue which I can address separately.

After applying the attached patch to ice40/bitstream.cc, block RAM clock edges behave correctly on the UP5K (presumably this is also true on other UP family members).
[bitstream.cc.diff.txt](https://github.com/YosysHQ/nextpnr/files/8010681/bitstream.cc.diff.txt)

You guys do fantastic work on such an incredible project!

bnossum, 2022-09-13:
I too believe there is something fishy with the SB_RAM40_4KNR for the UP5K. The testcase for an icebreaker board below should not light any LEDs in my opinion, but it does for the instance of SB_RAM40_4KNR. Compiled with
Yosys 0.21+7 (git sha1 d98738db5, gcc 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)
nextpnr-ice40 -- Next Generation Place and Route (Version nextpnr-0.3-89-gf1349e11)
```
/* For icebreaker. Assume 12 MHz clock 
  * 
 */
module top
  (
   input  CLK,
   output LED1,LED2,LED3,LED4,LED5
   );
   
   reg [15:0] cnt,cnt_1d,d0,d1;
   wire       clk;
   reg        rd0_error,any_rd0_error,rd1_error,any_rd1_error;
   reg [7:0]  startupcnt;

   always @(posedge CLK)
     if ( startupcnt[7] == 0 ) 
       startupcnt <= startupcnt+1;
   SB_GB clkbuf ( .USER_SIGNAL_TO_GLOBAL_BUFFER (CLK & startupcnt[7] ),
                  .GLOBAL_BUFFER_OUTPUT         (clk                 ) );
   
   always @(posedge clk) begin
      cnt    <= cnt+1;
      cnt_1d <= cnt;
   end
   
   SB_RAM40_4K # (.READ_MODE(0), .WRITE_MODE(0) ) works
     (
      .RDATA( d0                 ),
      .RADDR( {3'h0,cnt_1d[7:0]} ),
      .WADDR( {3'h0,cnt[7:0]}    ),
      .MASK(  16'h0              ),
      .WDATA( cnt                ),
      .RCLKE( 1'b1               ),
      .RCLK(  ~clk               ),
      .RE(    1'b1               ),
      .WCLKE( 1'b1               ),
      .WCLK(  clk                ),
      .WE(    1'b1               )
      );
   
   SB_RAM40_4KNR # (.READ_MODE(0), .WRITE_MODE(0) ) fails
     (
      .RDATA( d1                 ),
      .RADDR( {3'h0,cnt_1d[7:0]} ),
      .WADDR( {3'h0,cnt[7:0]}    ),
      .MASK(  16'h0              ),
      .WDATA( cnt                ),
      .RCLKE( 1'b1               ),
      .RCLKN( clk                ),
      .RE(    1'b1               ),
      .WCLKE( 1'b1               ),
      .WCLK(  clk                ),
      .WE(    1'b1               )
      );
   
    
   always @(posedge clk) begin
      rd0_error <= d0 != cnt_1d;
      any_rd0_error <= any_rd0_error | rd0_error;      
      rd1_error <= d1 != cnt_1d;
      any_rd1_error <= any_rd1_error | rd1_error;
   end
   
   assign LED1 = 0; // Unused red LED 
   
   assign LED2 = rd0_error;       
   assign LED3 = any_rd0_error;   
   
   assign LED4 = rd1_error;   
   assign LED5 = any_rd1_error;
   
endmodule
// Local Variables:
// verilog-library-files: ("~/GITSTUFF/FROM_GITHUB/yosys/share/ice40/cells_sim.v" )
// End:
```

jswrightoc, 2022-09-13:
I can observe the edge on which the data is changing looking at one of the read port bit lines routed to an I/O pad, along with the reference clock.  Without modification to how nextpnr assigns the configuration bits the transition is clearly happening on the wrong clock edge.  As soon as I have the opportunity I'll post a picture of the waveforms on my scope.

In addition, all instances of the SB_RAM40K primitive I am using on the UP5K are now working flawlessly.  Before the modification to nextpnr I would observe glitchy behavior consistent with setup & hold violations, but only under specific conditions.  Routing a signal of interest to an I/O pad for observation would usually make it behave itself.  Maddening stuff - I'm relieved to have definitively nailed this one down.

Glad this helped you get your project going!!  At the time I was trying to get a digital biquad filter up and running, and the glitches would cause it to become unstable and oscillate.  I didn't know if there was something wrong with my coefficients, the implementation using the SB_MAC16 block, or what.

The real fix is to swap the NR and NW config bits in the nextpnr source.  There's been a minor code restructuring since I created the patch file attached above (so it won't apply correctly), but the required fix is still the same.

bnossum, 2022-09-13:
It is bit worrisome that this issue has been open since February, without being tagged bug. However, it reminds me of one of the many reasons to stick to one flank of the clock.

gatecat, 2022-09-14:
Very sorry, this somehow slipped through the cracks. Can you perhaps double check that I applied the patch correctly in #1023 ?

PR #1023 (2022-09-16): ice40: Fix UltraPlus BRAM clock polarity
Fixes #910, using the patch from @jswrightoc
closed 2022-09-16 by gatecat