YosysHQ/nextpnr #911: AttributeError: module 'pytrellis' has no attribute 'load_database'

GarethH96, 2022-02-08:
OS: Arch Linux
Kernel: 5.16.5-arch1-1

NextPNR fails to build with ECP5 using the [prjtrellis](https://archlinux.org/packages/community/x86_64/prjtrellis/) and [prjtrellis-db](https://archlinux.org/packages/community/x86_64/prjtrellis-db/) packages provided by Arch's community repository. Lattice Diamond is not installed as I want to use a fully libre toolchain.

Output of `cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr -DTRELLIS_LIBDIR=/usr/lib/trellis && make -B`
```
-- The CXX compiler identification is GNU 11.1.0
-- The C compiler identification is GNU 11.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building with IPO
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.2", minimum required is "3.5") 
-- Found PythonLibs: /usr/lib/libpython3.10.so (found suitable version "3.10.2", minimum required is "3.5") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found version "1.78.0") found components: filesystem program_options iostreams system thread regex chrono atomic 
-- Found Boost: /usr/include (found version "1.78.0") found components: program_options filesystem system 
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr
-- Trellis library directory: /usr/lib/trellis
-- Trellis data directory: /usr/share/trellis
-- Using ECP5 chipdb: /usr/local/src/git/nextpnr/ecp5/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/git/nextpnr
[  1%] Generating chipdb/chipdb-25k.bba
[  3%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  5%] Linking CXX executable bbasm
Traceback (most recent call last):
  File "/usr/local/src/git/nextpnr/ecp5/trellis_import.py", line 640, in <module>
    main()
  File "/usr/local/src/git/nextpnr/ecp5/trellis_import.py", line 604, in main
    pytrellis.load_database(database.get_db_root())
AttributeError: module 'pytrellis' has no attribute 'load_database'
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:77: ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [CMakeFiles/Makefile2:255: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  5%] Built target bbasm
make: *** [Makefile:136: all] Error 2
```

Ganesh-Devdas, 2022-02-14:
Hi, I have got that issue on Debian/bookworm...

There seems to be more then one package called pytrellis ;)
You'd have to uninstall one spoiling the party :P

I have built from sources:
 - prjtrellis-1.1
 - nextpnr-0.1

However I had to link pytrellis.so into python3/dist-packages

GarethH96, 2022-02-15:
I started from scratch by uninstalling Trellis with `pip` and `yay`. Then, downloaded Trellis from [github](https://github.com/YosysHQ/prjtrellis), compiled, and installed with `/usr/local` as the prefix. Finally, I managed to build nextpnr without issue.

closed 2022-02-15 by GarethH96