YosysHQ/nextpnr #934: Question: force placement position of cells (ice40 --up5k)

Riesi, 2022-03-05:
Hello!
Is it possible to force the placement of cells of an existing design? 
I want to create multiple designs which only differ in the placement of the logic cells. I have the json file from Yosys and a constraints file. Can nextpnr-ice40 do such a thing?

I already looked over the documentation a bit, but I havent really found clues to that.

Ravenslofty, 2022-03-07:
You can use things like `BEL=` attributes for that; see [here](https://github.com/YosysHQ/nextpnr/blob/master/docs/constraints.md#absolute-placement-constraints).

Riesi, 2022-03-07:
That seems to be exactly what I need! I guess I overlooked this part of the docs. :grimacing: 
Thank you very much!

closed 2022-03-07 by Riesi