YosysHQ/nextpnr #935: icepll generated code unsupported

olell, 2022-03-06:
Hey there,
I'm experiencing an error which I am not able to resolve: I switched from arachne to nextpnr and tried to build an older project. I'm getting the following error when running nextpnr:
```
Info: Annotating ports with timing budgets for target frequency 12.00 MHz
ERROR: cell type 'pll' is unsupported (instantiated as 'video_clk_pll')
2 warnings, 1 error
```

The PLL module is generated by icepll and looks like this:
```
module pll(
        input  clock_in,
        output clock_out,
        output locked
        );

SB_PLL40_CORE #(
                .FEEDBACK_PATH("SIMPLE"),
                .DIVR(4'b0000),         // DIVR =  0
                .DIVF(7'b1000010),      // DIVF = 66
                .DIVQ(3'b101),          // DIVQ =  5
                .FILTER_RANGE(3'b001)   // FILTER_RANGE = 1
        ) uut (
                .LOCK(locked),
                .RESETB(1'b1),
                .BYPASS(1'b0),
                .REFERENCECLK(clock_in),
                .PLLOUTCORE(clock_out)
                );

endmodule
```
Using arachne again this works fine, maybe someone may help :D

gatecat, 2022-03-06:
Sounds like the `pll` module isn't being included in the design, can you provide more details on what you are running?

gatecat, 2022-08-18:
if this is still causing problems, please provide a bit more info & I'll reopen the issue

closed 2022-08-18 by gatecat