YosysHQ/nextpnr #959: ice40 sg48 pll pad pin 44 results in "not connected to any PLL BEL"

tboy32, 2022-03-21:
This is for a sg48 package.

pcf file:
```
...
set_io -nowarn ext_clk 44
...
```

verilog:
```
module top (input ext_clk);
...
    always @(posedge clk_48M) begin
    ...
    end

    pll_pad_48M pll_48M_instance (
        .clock_in(ext_clk),
        .clock_out(clk_48M),
        .locked()
    );
)

module pll_pad_48M(
        input  clock_in,
        output clock_out,
        output locked
    );
    
    SB_PLL40_PAD #( 
        .FEEDBACK_PATH("SIMPLE"),
        .DIVR(4'b0000),		// DIVR =  0
        .DIVF(7'b0111111),	// DIVF = 63
        .DIVQ(3'b100),		// DIVQ =  4
        .FILTER_RANGE(3'b001)	// FILTER_RANGE = 1
    ) uut (
        .LOCK(locked),
        .RESETB(1'b1),
        .BYPASS(1'b0),
        .PACKAGEPIN(clock_in),
        .PLLOUTCORE(clock_out)
    );
endmodule
```
Results in

`ERROR: PLL 'pll_48M_instance.uut' PACKAGEPIN SB_IO 'ext_clk$sb_io' is not connected to any PLL BEL`

Changing the pin number from 44 to 35 results in no error, however the external oscillator in my design is already connected to pin 44 so not easily changed. According to the datasheet both of these pins are Global inputs which can be used for clock nets. The only difference is pin 35 drives the GBUF0 buffer and pin 44 drives the GBUF6 buffer.  Is nextpnr missing pin 44 or am I missing a detail? Thanks.

gatecat, 2022-03-21:
Only pin 35 can drive the PLL input directly using a PAD type PLL. You can use a CORE type PLL instead and there will be a tiny bit of fabric routing but that shouldn't be a significant performance problem in most cases. However whenever using this type of PLL pin 35 can then only be used as an output (and also pin 37 for a dual output PLL).

tboy32, 2022-03-29:
Thanks for providing that information. I know this isn't an ice40 help forum so please excuse my questions (again I very much appreciate the help), but is there a document I missed that contains this information? If not I'm not sure how else I'd be able to identify & troubleshoot this issue besides submitting a request for assistance here. I'd also like to be aware of all such pin incompatibilities/limitations before designing a board with a particular FPGA pinout.

I asked Lattice about whether pin 44 could be used for a SB_PLL40_2_PAD primitive and they stated the following: 

> As stated in the document (ICE40 Technology Library), You may use pins found in the TOP and BOTTOM I/O Banks. Please do refer to the devices' pinout file to check available pins for BANKS 0 and 2 (https://www.latticesemi.com/view_document?document_id=51971).

According to the spreadsheet they referenced both pin 35 and 44 are shaded blue along with the other Global Gx pins. The only difference is pin 44 is on the Bottom (IOB) and pin 35 is on the Top (IOT). This response from Lattice makes me think this may be a nextpnr limitation and not necessarily one of the ice40. Might there be a another/better question I could ask Lattice to get a more complete understanding of where the limitation is?

gatecat, 2022-03-29:
To confirm, if you try to build this in icecube, you get the following failure:
```
PLL pll_48M_instance.uut Illegally placed at (6,0)
W2724: IO constraint specified on pin '44' is infeasible. Ignoring the constraint
```
and the constraint is just ignored.

Unfortunately the documentation for this isn't currently great anywhere. The Lattice docs seem to be based on the 8k which has both top and bottom PLLs and not the 5k which has just a top PLL.

Are you sure a `CORE` type PLL isn't alright for your use case?

gatecat, 2022-08-18:
As this design fails in icecube too, not a nextpnr specific issue but an ice40 hardware limit.

closed 2022-08-18 by gatecat