YosysHQ/nextpnr #967: Ignore clock-domain crossing synchronizer paths

basusamit, 2022-03-30:
HI all!

First off, thank you for making such an awesome set of tools!  NextPNR and Yosys are very, very nice to work with.  

One issue I have run into with one of my designs for an ECP5 board is the need to ignore setup/hold constraints on a path/net that crosses between clock domains.  I have an asynchronous FIFO and where the clock domains are crossed (using a flag-based synchronizer), `nextpnr` is unhappy about the timing.  Normally, I would think one would add a constraint telling `nextpnr` to ignore the timing on that path.  But I could not see how to do so.

So, my question is:
1.  Is there a way to tell `nextpnr` to ignore timing on a net?  (I did review the docs and the previous issues).

2. If not, is this a useful feature?

3. If it's useful, could someone sketch out a few notes about how to do it?  I'm willing to take a shot at adding the capability.

Thanks!
Samit

rowanG077, 2022-04-12:
I'm interested in this as well. I would be willing to work together with you. Honestly multi clock designs feel like you are on very thin ice with nextpnr...

smunaut, 2022-09-19:
Huh, AFAIK there is no cross-clock constraining at all ATM so in effect, all paths are ignored ... so you'd need to add support for constraining them first before you can ignore some.