YosysHQ/nextpnr #979: nextpnr generic architecture seg faults when unbinding a wire

AbedFares, 2022-04-26:
I've just run simple.sh located under generic/examples. it works fine! Then I tried to unbind a wire in the postRoute script but it seg faults in line 390 in generic/arch.cc
I suspect that there is a missing condition on whether or not the "uarch" variable is nullptr or not. I tried adding this condition and it passed.
Attached is where exactly the condition should be and the seg fault that I've received.
[image]

gatecat, 2022-04-27:
Thanks for reporting, can you confirm that 20cfafa1 solves this?

AbedFares, 2022-04-27:
Thanks for the reply. Yes, I confirm that it was solved!

closed 2022-04-27 by AbedFares