YosysHQ/nextpnr #986: nextpnr-ice40: error while loading shared libraries: libtbb.so.2: cannot open shared object file: No such file or directory

cpt-n3mo, 2022-05-19:
good day,

nextpnr will not run with the following error 
``` bash
nextpnr-ice40: error while loading shared libraries: libtbb.so.2: cannot open shared object file: No such file or directory
```
Please update libtbb.so.2  to current version. libtbb.so.12

regards

gatecat, 2022-05-19:
Have you tried a clean rebuild of nextpnr?

cpt-n3mo, 2022-05-19:
> Have you tried a clean rebuild of nextpnr?

yes i have.

cpt-n3mo, 2022-05-19:
sorry
im running arch linux 5.15.40-1-lts
gcc version 12.1.0 (GCC)
cmake version 3.23.1
nextpnr-ice40-nightly from AUR
https://github.com/YosysHQ/nextpnr.git"#commit=4ecbf6c6

rebuilding todays update... i will report back with some more info when rebuild is done..

cpt-n3mo, 2022-05-19:
here is output of ldd

``` bash
        linux-vdso.so.1 (0x00007ffc1ad46000)
	libtbb.so.2 => not found
	libboost_filesystem.so.1.76.0 => not found
	libboost_program_options.so.1.76.0 => not found
	libboost_iostreams.so.1.76.0 => not found
	libboost_system.so.1.76.0 => not found
	libboost_thread.so.1.76.0 => not found
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f511ce7a000)
	libboost_regex.so.1.76.0 => not found
	libboost_chrono.so.1.76.0 => not found
	libboost_date_time.so.1.76.0 => not found
	libboost_atomic.so.1.76.0 => not found
	libpython3.9.so.1.0 => not found
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f511cc41000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f511cb5a000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f511cb3a000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f511c92d000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007f512ac76000)
```

cpt-n3mo, 2022-05-19:
it seems package lib32-tbb containing libtbb.so.2 is corrupted and will not install anymore..
trying to rebuild lib32-tbb failed.. will have to go down stream for this one i guess.

gatecat, 2022-05-19:
what about running `which cmake`?

cpt-n3mo, 2022-05-19:
> what about running `which cmake`?

outputs
``` bash
/usr/bin/cmake
```

gatecat, 2022-05-19:
most strange, the problem seems to be cmake picking the wrong version of tbb, and based on the ldd output other libraries too, rather than libtbb.so.2 being missing per se

gatecat, 2022-05-19:
if you just want to use nextpnr, rather than build it, at this stage my recommendation is just to get the prebuilt binaries from https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2022-05-19

cpt-n3mo, 2022-05-19:
> if you just want to use nextpnr, rather than build it, at this stage my recommendation is just to get the prebuilt binaries from https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2022-05-19

nice ! this works like a charm!
this is quite nice actually. :)

i will have to take a look why my system is not getting the correct version, for now i can continue with my project ! major thanks for your super quick response !

kind regards

closed 2022-05-19 by cpt-n3mo