YosysHQ/nextpnr #994: Crosslink NX evaluation board fails nextpnr

jaycbhsn, 2022-06-02:


gatecat, 2022-06-02:
Thanks, definitely a bug but won't be able to look into it for a day or two, most likely.

gatecat, 2022-07-19:
Sorry for the delay, it's been a hectic couple of months, can you see if https://github.com/gatecat/prjoxide/commit/c2186112a678d20c67547cf72f8d43c704b9dad4 fixes this (first rebuild prjoxide with that commit, then rebuild nextpnr so it gets the updated device data)?

jaycbhsn, 2022-07-22:
Hello!
I built with the new patch, but I'm getting another error (I believe this is seperate. I probably changed something since I ran last time):
Info:     constraining PLL_0 'PLL_CORE' to bel 'X1/Y0/PLL_ULC'
terminate called after throwing an instance of 'nextpnr_nexus::assertion_failure'
  what():  Assertion failure: start_wire != WireId() (/work/_builds/linux-x64/nextpnr-nexus/nextpnr/nexus/pack.cc:689)
build_lattice_crosslink_nx_evn.sh: line 4: 16973 Aborted                 (core dumped) nextpnr-nexus --json lattice_crosslink_nx_evn.json --pdc lattice_crosslink_nx_evn.pdc --fasm lattice_crosslink_nx_evn.fasm --device LIFCL-40-8BG400CES --timing-allow-fail --seed 1
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/jacobqh/litex-boards/litex_boards/targets/lattice_crosslink_nx_evn.py", line 125, in <module>
    main()
  File "/home/jacobqh/litex-boards/litex_boards/targets/lattice_crosslink_nx_evn.py", line 118, in main
    builder.build(**builder_kargs)
  File "/home/jacobqh/litex/litex/soc/integration/builder.py", line 350, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/jacobqh/litex/litex/soc/integration/soc.py", line 1208, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/jacobqh/litex/litex/build/lattice/platform.py", line 43, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/jacobqh/litex/litex/build/lattice/oxide.py", line 169, in build
[log.txt](https://github.com/YosysHQ/nextpnr/files/9169665/log.txt)

    _run_script(script)
  File "/home/jacobqh/litex/litex/build/lattice/oxide.py", line 104, in _run_script
    raise OSError("Error occured during Yosys/Nextpnr's script execution.")
OSError: Error occured during Yosys/Nextpnr's script execution.

And again, here is the log file:

gatecat, 2022-08-18:
Did you rebuild nextpnr after rebuilding prjoxide?

closed 2022-09-14 by gatecat