YosysHQ/nextpnr #1003: SB_WARMBOOT has no effect on iCE40-LP384

machdyne, 2022-07-14:
Hello. I'm wondering if anyone has had any success getting warm boot to work on an iCE40-LP384 or can definitively tell me whether or not the LP384 supports warm boot. Nextpnr says that it is supported when I build gateware for LP384.

I've tried the same simple gateware on an HX4K and an LP384 and triggering warm boot works fine on the HX4K but seems to have no effect on the LP384.

Thanks.

gatecat, 2022-07-20:
Testing a design using WARMBOOT on the 384 through icecube seems that it isn't supported:

```
Error: Current device or package (iCE40LP384-CM49) does not support SB_WARMBOOT. Please change the family/device/package and re-import your design.
```

I'm not sure why it is in the icestorm db, or why the Lattice docs don't really mention this either. I will see if I can find anything else out but at this point best to assume it's just missing or broken in the hw :/

machdyne, 2022-07-20:
Thanks for the response. Lattice support has told me that:

"SB_WARMBOOT primitive is not supported for iCE40-LP384 device."

But after some back-and-forth it's still not clear if that means software or hardware support.

I found nothing mentioning the lack of support in any of the datasheets, only that the LP384 doesn't have a PLL.

I have designed a board [Winzig](https://github.com/machdyne/winzig) with the LP384 that would benefit from a bootloader but I am assuming at this point that it's not gonna happen.

Thanks again.

closed 2023-09-13 by gatecat