YosysHQ/nextpnr #1007: nexus: Design appears to not meet timing when nextpnr thinks it does

davidlattimore, 2022-08-02:
We have a design that exhibits non-deterministic behaviour when run at a high clock speed (div=5), but behaves correctly at a lower clock speed (div=6).
We were able to [fix this by breaking a path](https://github.com/google/CFU-Playground/commit/f323e5840ce06214b1e69abdeb1c7399ff210c32): LRAM, addition, multiply-accumulator.
Our guess is that the timing requirements of either the output of the LRAM or the input of the multiply-accumulator might be incorrect.
The FPGA we’re using is a CrossLink Nexus LIFCL-17-7UWG72C.
We’re using a period constraint 7% tighter than our actual clock frequency, since the CrossLink-NX internal oscillator runs at +/- 7% of nominal frequency.
I’ve attached our build directory for 3 different configurations:
* [hps-build-div-5-with-fix.zip](https://github.com/YosysHQ/nextpnr/files/9238288/hps-build-div-5-with-fix.zip): Gives correct, deterministic results.
* [hps-build-div-5-without-fix.zip](https://github.com/YosysHQ/nextpnr/files/9238291/hps-build-div-5-without-fix.zip): Gives incorrect, non-deterministic results.
* [hps-build-div-6-without-fix.zip](https://github.com/YosysHQ/nextpnr/files/9238294/hps-build-div-6-without-fix.zip): Gives correct, deterministic results.

The verilog produced by Amaranth is included as build/cfu.v.
The final bitstream is at build/hps_platform/gateware/hps_platform.bit.
Versions:
* nextpnr-nexus -- Next Generation Place and Route (Version 664cec54)
prjoxide 0.1
    * Commit d1fc5cd045531a0cc48ebffd5802f4ef84067aea
* Yosys 0.19+18 (git sha1 08c319fc352, x86_64-pc-linux-gnu-clang++ 15.0_pre458507_p20220602-r16 -O2 -fPIC -Os)
    * Has a revert of https://github.com/YosysHQ/yosys/commit/0a8eaca322a622610efe5a0d33d1cda83e4afa8c due to https://github.com/YosysHQ/yosys/issues/3416

gatecat, 2022-08-03:
This does sound like a believeable problem that I will investigate, but just one question if you have time to  - if you try making other small design changes (even the `--seed` parameter to nextpnr), how reliable are the passes or failures in each configuration?

davidlattimore, 2022-08-03:
Thanks @gatecat! Since implementing the fix and reducing our clock divisor to 5, we've made a number of unrelated changes to our design. With each change, we've generally needed to change the seed that we use in order to successfully meet timing. With those changes, the bug has not reoccurred. So I believe the passes to reliable.

I just tried a few different seeds and a design change with the failing design and all failed, so I think that's consistent as well.

gatecat, 2022-08-03:
Thanks, that's very useful info.

gatecat, 2022-08-10:
There seem to be two separate issues going on here, if I am understanding correctly.

The first is that the timing data was simply missing for LRAM. This was a relatively easy fix in, https://github.com/gatecat/prjoxide/pull/28 and https://github.com/YosysHQ/nextpnr/pull/1011 which also updates the timing data to the production Radiant while we're at it.

Unfortunately, this doesn't solve the issue at hand because the relevant path appears to be a cross-domain one between `glb_clk` and `cfu_clk`. nextpnr simply doesn't support constraining these on a pass/fail basis at the moment, whilst it's on the TODO list I can't give an eta for that as it would be part of a general significant upgrade of the timing analysis rather than just an ad hoc patch(*). However, the report does show how slow the LRAM output is which was almost certainly the underlying problem here:

```
Info: Critical path report for cross-domain path 'posedge clkout$glb_clk' -> 'posedge cfu_clk':
Info: curr total
Info: 12.7 12.7  Source DPSC512K_1.DOB20
Info:  3.4 16.1    Net cfu_lram_bus_lram1_din[20] budget -1.934000 ns (74,15) -> (10,11)
Info:                Sink Cfu.core.f0_data_out0_LUT4_Z_11_C_LUT4_Z_D_WIDEFN9_Z$widefn_comb[0]$.B
Info:                Defined in:
Info:                  /usr/local/google/home/dml/chromiumos/src/platform/hps-firmware2/build/hps_platform/gateware/hps_platform.v:124.13-124.28
Info:  0.3 16.4  Source Cfu.core.f0_data_out0_LUT4_Z_11_C_LUT4_Z_D_WIDEFN9_Z$widefn_comb[0]$.OFX
Info:  0.7 17.1    Net Cfu.core.f0_data_out0_LUT4_Z_11_C_LUT4_Z_D budget -0.967000 ns (10,11) -> (13,7)
Info:                Sink Cfu.core.f0_data_out0_LUT4_Z_11_C_LUT4_Z.D
Info:  0.2 17.3  Source Cfu.core.f0_data_out0_LUT4_Z_11_C_LUT4_Z.F
Info:  0.4 17.7    Net Cfu.core.f0_data_out0_LUT4_Z_11_C budget -0.967000 ns (13,7) -> (12,7)
Info:                Sink Cfu.core.val_0_0_CCU2_S0_2_A0_LUT4_Z.D
Info:  0.2 17.9  Source Cfu.core.val_0_0_CCU2_S0_2_A0_LUT4_Z.F
Info:  0.9 18.9    Net Cfu.core.val_0_0_CCU2_S0_2_A0 budget -0.967000 ns (12,7) -> (11,18)
Info:                Sink Cfu.core.val_0_0_CCU2_S0_2$ccu2_comb[0]$.A
Info:  0.3 19.2  Source Cfu.core.val_0_0_CCU2_S0_2$ccu2_comb[0]$.FCO
Info:  0.0 19.2    Net Cfu.core.val_0_0_CCU2_S0_2$widefn_int_cy$ budget -0.488000 ns (11,18) -> (11,18)
Info:                Sink Cfu.core.val_0_0_CCU2_S0_2$ccu2_comb[1]$.FCI
Info:  0.0 19.2  Source Cfu.core.val_0_0_CCU2_S0_2$ccu2_comb[1]$.FCO
Info:  0.1 19.3    Net Cfu.core.val_0_0_CCU2_S0_2_COUT budget -0.447000 ns (11,18) -> (12,18)
Info:                Sink Cfu.core.val_0_0_CCU2_S0_1$ccu2_comb[0]$.FCI
Info:                Defined in:
Info:                  /usr/local/google/home/dml/chromiumos/src/platform/hps-firmware2/build/hps_platform/gateware/hps_platform.v:2142.5-2161.2
Info:                  /usr/local/google/home/dml/src/CFU-Playground/proj/hps_accel/gateware/gen2/accelerator.py:267
Info:                  /usr/local/google/home/dml/chromiumos/src/platform/hps-firmware2/soc/cfu.v:403.8-442.4
Info:                  /usr/bin/../share/yosys/nexus/arith_map.v:83.5-89.4
Info:                  /usr/bin/../share/yosys/nexus/cells_sim.v:201.26-201.30
Info:  0.0 19.3  Source Cfu.core.val_0_0_CCU2_S0_1$ccu2_comb[0]$.FCO
Info:  0.0 19.3    Net Cfu.core.val_0_0_CCU2_S0_1$widefn_int_cy$ budget -0.413000 ns (12,18) -> (12,18)
Info:                Sink Cfu.core.val_0_0_CCU2_S0_1$ccu2_comb[1]$.FCI
Info:  0.2 19.5  Source Cfu.core.val_0_0_CCU2_S0_1$ccu2_comb[1]$.F
Info:  0.9 20.4    Net Cfu.core.sysarray_input_a0[7] budget -0.413000 ns (12,18) -> (7,19)
Info:                Sink Cfu.core.sysarray.macc_0_0.dsp/MULT9_CORE_x0_z2.A7
Info:                Defined in:
Info:                  /usr/local/google/home/dml/chromiumos/src/platform/hps-firmware2/build/hps_platform/gateware/hps_platform.v:2142.5-2161.2
Info:                  /usr/local/google/home/dml/src/CFU-Playground/proj/hps_accel/gateware/gen2/accelerator.py:266
Info:                  /usr/local/google/home/dml/chromiumos/src/platform/hps-firmware2/soc/cfu.v:403.8-442.4
Info:  1.2 21.7  Source Cfu.core.sysarray.macc_0_0.dsp/MULT9_CORE_x0_z2.P189
Info:  0.7 22.3    Net Cfu.core.sysarray.macc_0_0.dsp/MULT9_CORE_x0_z2$conn$P189 budget -0.383000 ns (7,19) -> (9,19)
Info:                Sink Cfu.core.sysarray.macc_0_0.dsp/REG18_CORE_x2_z0.PM9
Info:  0.6 22.9  Setup Cfu.core.sysarray.macc_0_0.dsp/REG18_CORE_x2_z0.PM9
Info: 16.0 ns logic, 7.0 ns routing
```
(*) I might be willing to do an ad hoc patch on a funded basis for this specific case of a gated clock. It is also worth noting that the DCC will introduce a phase shift on the clock as well that won't be analysed here at the moment...

davidlattimore, 2022-08-26:
Thanks @gatecat!

The timing for DOB20 is certainly quite high. I had a bit of a look into the timing the LRAM outputs and it seems that whether it's registered or not has a very large effect. 11775 ps for non-registered vs 1616 ps for registered. I've sent a question to Lattice to confirm if that's correct and the difference really is that large.

Whether the output is registered can be changed independently for each port.

```verilog
DPSC512K #(
      //.OUTREG_A("OUT_REG"),
      .OUTREG_B("OUT_REG")
    ) lram_0 (
      ...
    );
```

Is there any precedent in nextpnr for deciding timing based on properties like this?

mkurc-ant, 2022-08-31:
Hi guys. I've prepared a proof-of-concept implementation of cross-domain timing checks in nextpnr. Please take a look: https://github.com/YosysHQ/nextpnr/pull/1019 I've tested this on the attached designs and since all the gated clocks come from the same source it behaves as if they were a single clock.

For the `hps-build-div-5-without-fix` example I have in the report:
```
Info: Max frequency for posedge cfu_clk        -> posedge clkout$glb_clk: 109.91 MHz (PASS at 96.30 MHz)
Info: Max frequency for posedge cfu_clk        -> posedge sdrio_clk     : 120.80 MHz (PASS at 96.30 MHz)
ERROR: Max frequency for posedge clkout$glb_clk -> posedge cfu_clk       : 45.59 MHz (FAIL at 96.30 MHz)
ERROR: Max frequency for posedge clkout$glb_clk -> posedge sdrio_clk     : 59.36 MHz (FAIL at 96.30 MHz)
Info: Max frequency for posedge sdrio_clk      -> posedge cfu_clk       : 119.50 MHz (PASS at 96.30 MHz)
ERROR: Max frequency for posedge sdrio_clk      -> posedge clkout$glb_clk: 96.09 MHz (FAIL at 96.30 MHz)
```

And for `hps-build-div-6-without-fix`:
```
Info: Max frequency for posedge cfu_clk        -> posedge clkout$glb_clk: 103.43 MHz (PASS at 80.26 MHz)
Info: Max frequency for posedge cfu_clk        -> posedge sdrio_clk     : 100.36 MHz (PASS at 80.26 MHz)
ERROR: Max frequency for posedge clkout$glb_clk -> posedge cfu_clk       : 43.43 MHz (FAIL at 80.26 MHz)
ERROR: Max frequency for posedge clkout$glb_clk -> posedge sdrio_clk     : 54.85 MHz (FAIL at 80.26 MHz)
Info: Max frequency for posedge sdrio_clk      -> posedge cfu_clk       : 105.65 MHz (PASS at 80.26 MHz)
Info: Max frequency for posedge sdrio_clk      -> posedge clkout$glb_clk: 87.27 MHz (PASS at 80.26 MHz)
```

Both of them fail timing due to the recently merged PR with LRAM timings - propagation delays are quite significant there.

gatecat, 2022-08-31:
Thanks, this looks very promising at a glance! a detailed review will take a few days as I've been busy with FPL stuff but do poke me if I don't get back within a week.

mkurc-ant, 2022-09-01:
@gatecat Thanks. One thing I'd like you to check more thoroughly is the way of identifying common clock drivers. For example I know that it would give false positives when there is a clock buffer/mux with a control input that has a combinational timing relation with the clock output.

rowanG077, 2022-09-07:
Does the same issue hold for any clock crossing RAM or is nexus LRAM specific?

closed 2023-09-13 by gatecat