YosysHQ/nextpnr #1009: Master fails to build due to argument mismatch in pytrellis.so

shareefj, 2022-08-06:
I've installed the latest version of Project Trellis from master and have been trying to build `nextpnr` but am hitting the following issue:

```
[  3%] Generating chipdb/chipdb-25k.bba
[  3%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
Traceback (most recent call last):
  File "/home/shareefj/git/nextpnr/ecp5/trellis_import.py", line 701, in <module>
    main()
  File "/home/shareefj/git/nextpnr/ecp5/trellis_import.py", line 689, in main
    ddrg = pytrellis.make_dedup_chipdb(chip, include_lutperm_pips=True, split_slice_mode=True)
TypeError: make_dedup_chipdb(): incompatible function arguments. The following argument types are supported:
    1. (chip: pytrellis.Chip, include_lutperm_pips: bool = False) -> pytrellis.DedupChipdb

Invoked with: <pytrellis.Chip object at 0x7f0b6ffe67b0>; kwargs: include_lutperm_pips=True, split_slice_mode=True
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:77: ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [CMakeFiles/Makefile2:255: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

This seems to be due to the `split_slice_mode=True` argument not being expected.  However, I can see the relevant commit in Project Trellis where this change was made so I'm not sure where the build issue is coming from.

Any thoughts?

gatecat, 2022-08-07:
Are you sure nextpnr is being pointed to the same, latest version of trellis that you have pulled and built? For example, that you installed the new trellis that you built.

shareefj, 2022-08-07:
Yes, I'm installing to /opt and my cmake command for Project Trellis is `cmake -DCMAKE_INSTALL_PREFIX=/opt/prjtrellis/master .` and my subsequent command for nextpnr is `cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/opt/prjtrellis/master`.  I also know it's picking it up correctly as I managed to forget the `make install` step for Trellis when I moved from the 1.2.1 tag to master and got a different error because nothing existed at `/opt/prjtrellis/master`.

shareefj, 2022-08-07:
So the issue seems to be that either your instructions aren't complete or the way CMake caches paths needs to be documented.  I have no CMake experience but noticed that I was getting different paths printed when running cmake:

```
$ cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/opt/prjtrellis/master
-- Building with IPO
-- Found Boost: /usr/include (found version "1.74.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic
-- Found Boost: /usr/include (found version "1.74.0") found components: program_options filesystem system
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /opt/prjtrellis/master
-- Trellis library directory: /opt/prjtrellis/1.2.1/lib/trellis
-- Trellis data directory: /opt/prjtrellis/master/share/trellis
-- Using ECP5 chipdb: /home/shareefj/git/nextpnr/ecp5/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /home/shareefj/git/nextpnr
```

The file `CMakeCache.txt` contained a path that needs to be overwritten when rebuilding a new tag/release/branch: `TRELLIS_LIBDIR:PATH=/opt/prjtrellis/1.2.1/lib/trellis`.

gatecat, 2022-08-18:
Unfortunately I think this just has to go down as one of those CMake things...

closed 2022-08-18 by gatecat