YosysHQ/nextpnr #1020: compile file for nextpnr

steven07210011, 2022-09-06:
when i type command in raspberrypi to compile nextpnr:

pi@raspberrypi:~/Projects/fpga/nextpnr $ cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local -DCMAKE_INSTALL_PREFIX=/usr/local -DBUILD_GUI=on .
it show
-- The CXX compiler identification is GNU 10.2.1
-- The C compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building with IPO
-- Found Python3: /usr/bin/python3.9 (found suitable version "3.9.2", minimum required is "3.5") found components: Interpreter 
-- Found Python3: /usr/include/python3.9 (found suitable version "3.9.2", minimum required is "3.5") found components: Development Development.Module Development.Embed 
CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

mmicko, 2022-09-07:
You are probably missing boost dev libraries. Take a look at https://github.com/YosysHQ/nextpnr#prerequisites for details about which packages are needed to be installed to be able to compile nextpnr.

steven07210011, 2022-09-07:
> You are probably missing boost dev libraries. Take a look at https://github.com/YosysHQ/nextpnr#prerequisites for details about which packages are needed to be installed to be able to compile nextpnr.

u means that i need to run command sudo apt install libboost-all-dev in nextpnr directory?

mmicko, 2022-09-07:
That is system wide operation, and it will install development libraries on your raspberrypi, so it is irrelevant from which place you execute it.

steven07210011, 2022-09-07:
pi@raspberrypi:~/Projects/fpga/nextpnr $ sudo apt install libboost-dev libboost-filesystem-dev libboost-thread-dev libboost-program-options-dev libboost-iostreams-dev libboost-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libboost-dev is already the newest version (1.74.0.3+b1).
libboost-filesystem-dev is already the newest version (1.74.0.3+b1).
libboost-iostreams-dev is already the newest version (1.74.0.3+b1).
libboost-program-options-dev is already the newest version (1.74.0.3+b1).
libboost-thread-dev is already the newest version (1.74.0.3+b1).
The following packages were automatically installed and are no longer required:
  erlang-base erlang-crypto erlang-syntax-tools libaubio5 libfuse2 libgit2-1.1 libhttp-parser2.9 libmbedcrypto3 libmbedtls12 libmbedx509-0 liboscpack1 libportaudio0 librtaudio6 librtmidi5 libruby2.7 libscsynth1 libsctp1 libsox-fmt-alsa
  libsox-fmt-base libsox3 libstk-4.6.1 osmid pulseaudio-module-jack rake ruby ruby-activesupport ruby-atomic ruby-aubio ruby-concurrent ruby-ffi ruby-hamster ruby-i18n ruby-kramdown ruby-memoist ruby-minitest ruby-multi-json
  ruby-net-telnet ruby-oj ruby-polyglot ruby-power-assert ruby-rouge ruby-rubame ruby-rubygems ruby-rugged ruby-sys-proctable ruby-test-unit ruby-thread-safe ruby-treetop ruby-tzinfo ruby-wavefile ruby-websocket ruby-xmlrpc
  ruby-zeitwerk ruby2.7 rubygems-integration sonic-pi-samples sox
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 19 not upgraded.
pi@raspberrypi:~/Projects/fpga/nextpnr $  sudo apt install libboost-all-dev 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libboost-all-dev is already the newest version (1.74.0.3+b1).
The following packages were automatically installed and are no longer required:
  erlang-base erlang-crypto erlang-syntax-tools libaubio5 libfuse2 libgit2-1.1 libhttp-parser2.9 libmbedcrypto3 libmbedtls12 libmbedx509-0 liboscpack1 libportaudio0 librtaudio6 librtmidi5 libruby2.7 libscsynth1 libsctp1 libsox-fmt-alsa
  libsox-fmt-base libsox3 libstk-4.6.1 osmid pulseaudio-module-jack rake ruby ruby-activesupport ruby-atomic ruby-aubio ruby-concurrent ruby-ffi ruby-hamster ruby-i18n ruby-kramdown ruby-memoist ruby-minitest ruby-multi-json
  ruby-net-telnet ruby-oj ruby-polyglot ruby-power-assert ruby-rouge ruby-rubame ruby-rubygems ruby-rugged ruby-sys-proctable ruby-test-unit ruby-thread-safe ruby-treetop ruby-tzinfo ruby-wavefile ruby-websocket ruby-xmlrpc
  ruby-zeitwerk ruby2.7 rubygems-integration sonic-pi-samples sox
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 19 not upgraded.

this means that i already installed?

steven07210011, 2022-09-07:
> That is system wide operation, and it will install development libraries on your raspberrypi, so it is irrelevant from which place you execute it.

pi@raspberrypi:~/Projects/fpga/nextpnr $ cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local .
-- The CXX compiler identification is GNU 10.2.1
-- The C compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building with IPO
-- Found Python3: /usr/bin/python3.9 (found suitable version "3.9.2", minimum required is "3.5") found components: Interpreter 
-- Found Python3: /usr/include/python3.9 (found suitable version "3.9.2", minimum required is "3.5") found components: Development Development.Module Development.Embed 
CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  CMakeLists.txt:184 (find_package)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
[... 53 lines cut ...]

-- Found Boost: /usr/include (found version "1.74.0") found components: program_options filesystem system 
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Searching 16 bit integer - Using unsigned short
-- Check if the system is big endian - little endian
-- Configuring architecture: ice40
-- Enabled iCE40 devices: 384;1k;5k;u4k;8k
-- Found Python3: /usr/bin/python3.9 (found suitable version "3.9.2", minimum required is "3.5") found components: Interpreter 
-- IceStorm install prefix: /usr/local
-- icebox data directory: /usr/local/share/icebox
-- Using iCE40 chipdb: /home/pi/Projects/fpga/nextpnr/ice40/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Projects/fpga/nextpnr

after i install, still got same error which are (-- Check for working CXX compiler: /usr/bin/c++ - skipped )and -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed

steven07210011, 2022-09-07:
> This is fine, it just points out that threading library is not available. You now have make files generated, so need to build it.

means that i can run the command make -j 4?

mmicko, 2022-09-07:
right

mmicko, 2022-09-07:
Yes. Those are not errors, but info about of cmake searching for tools/libraries and are perfectly normal. If makefiles are generated, as stated in output all is ready to be built.

closed 2022-09-07 by mmicko