YosysHQ/nextpnr #1022: ERROR DURING COMMAND MAKE ICESTICK

steven07210011, 2022-09-07:
Info: Program finished normally.
icetime -p BOARDS/icestick.pcf -P tq144 -r femtosoc.timings -d hx1k -t femtosoc.asc 
// Reading input .pcf file..
// Reading input .asc file..
// Reading 1k chipdb file..
// Creating timing netlist..
// Timing estimate: 19.56 ns (51.12 MHz)
icepack -s femtosoc.asc femtosoc.bin
iceprog femtosoc.bin
init..
Can't find iCE FTDI USB device (vendor_id 0x0403, device_id 0x6010 or 0x6014).
ABORT.
make: *** [BOARDS/icestick.mk:21: ICESTICK.prog] Error 2
pi@raspberrypi:~/Projects/fpga/learn-fpga/FemtoRV $ 

it show this error, how can i solve it?

rowanG077, 2022-09-07:
This is the wrong issue tracker. `iceprog` is part of  `icestorm` not `nextpnr`. But to answer your question it can't find the right programmer USB device as stated in the error.

steven07210011, 2022-09-07:
> This is the wrong issue tracker. `iceprog` is part of `icestorm` not `nextpnr`. But to answer your question it can't find the right programmer USB device as stated in the error.

its means cannnot detact the ice40 stick? or the ftdi friver wrong? since i did not have ice stick, i did not plug in it, will it lead to this error?

steven07210011, 2022-09-07:
> This is the wrong issue tracker. `iceprog` is part of `icestorm` not `nextpnr`. But to answer your question it can't find the right programmer USB device as stated in the error.

or it only can run with stick?

closed 2022-09-07 by steven07210011