YosysHQ/nextpnr #1027: Ice40 UP5K: SB_MAC16 Carry Out inoperative

jswrightoc, 2022-09-17:
The SB_MAC16 primitive located at tiles (25,10) thru (25,13) does not appear have a functioning carry out (CO). According to icestorm documentation this output is available on the IPConnect tile above at (25,14) as slf_op_0 and is routable to general-purpose logic.

CO may be mapped to a different location for this SB_MAC16 primitive, or perhaps it is simply not available (e.g. silicon errata).  The official Lattice documentation doesn't say much about CI and CO, and different documents are inconsistent with their description of CI and CO.  I do not know if the Lattice tools allow utilization of of the carry signals.

I have observed this on a Lattice Ice40 UltraPlus Breakout Board as well as several of my own designs, always with the same SB_MAC16 located at these tiles.  Instantiating all 8 SB_MAC16 primitives as multiply-accumulate with non-zero A and B inputs, and routing CO of each one to I/O pads yields 7 functional, and one always low.  I presume the behavior is the same regardless of configuration (MAC signed, MAC unsigned, accumulate only, etc.) as long as the accumulator is involved.

gatecat, 2022-09-19:
Huh, this is an interesting one. Some poking about with icecube does suggest that it happily routes that carry out via `(25, 14, 'slf_op_0')` as the other DSPs. I'll have to set up some hardware tests...

jswrightoc, 2022-09-19:
I can make a more detailed analysis or run specific hardware tests if that's helpful.  I have a Lattice UP5K breakout board and a scope.

I seldom use all 8 DSPs on the device (usually run out of logic cells first), is there a quick and easy way to never allocate the SB_MAC16 in question?

gatecat, 2022-09-20:
It would be very helpful for you to provide the mentioned design you're using to test?

jswrightoc, 2022-09-20:
Sorry, I wanted to at least make the issue known.  Had to verify a couple more things before I submitted the test design.

[up5k-breakout.pcf.txt](https://github.com/YosysHQ/nextpnr/files/9610551/up5k-breakout.pcf.txt)
[ice40.v.txt](https://github.com/YosysHQ/nextpnr/files/9610552/ice40.v.txt)
[ice40.asc.txt](https://github.com/YosysHQ/nextpnr/files/9611152/ice40.asc.txt)

Built with flags: --up5k --package sg48 --freq 24.000 --opt-timing --pcf-allow-unconstrained

In this particular configuration the inoperative carry output shows up on TEST0_OUT / pin 4 / 8A.

Any possible workaround as previously mentioned?

gatecat, 2022-09-26:
Sorry for the delay, hopefully this is sorted with https://github.com/YosysHQ/nextpnr/pull/1030

jswrightoc, 2022-09-26:
I confirm that works for the test case; I expect it will work in other designs as well.  Thank you for taking care of this!

closed 2022-09-26 by jswrightoc