YosysHQ/nextpnr #1038: ECP5 is not generating any output

Chandler-Kluser, 2022-10-28:
I am trying to place and route a simple nand module for `LFE5U-25F-6BG256C`:

top.v:
```
module top(
    input a,b,
    output c
);
nand(c,a,b);
endmodule
```
top.lpf:
```
LOCATE COMP "a" SITE "E1";
LOCATE COMP "b" SITE "F2";
LOCATE COMP "c" SITE "C2";

IOBUF PORT "a" IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "b" IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "c" IO_TYPE=LVCMOS33 DRIVE=4;
```
Running Yosys 0.22+57 (build from source in commit sha1 518194fac):
```
$ yosys top.v -p "synth_ecp5 -top top -json top.json;"
```
which gave me the following output:
```
/----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.22+57 (git sha1 518194fac, clang 14.0.0-1ubuntu1 -fPIC -Os)

-- Parsing `top.v' using frontend ` -vlog2k' --

1. Executing Verilog-2005 frontend: top.v
Parsing Verilog input from `top.v' to AST representation.
Storing AST representation for module `$abstract\top'.
Successfully finished Verilog frontend.

-- Running command `synth_ecp5 -top top -json top.json;' --

2. Executing SYNTH_ECP5 pass.

2.1. Executing Verilog-2005 frontend: /usr/local/bin/../share/yosys/ecp5/cells_sim.v
Parsing Verilog input from `/usr/local/bin/../share/yosys/ecp5/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\LUT4'.
Generating RTLIL representation for module `\$__ABC9_LUT5'.
Generating RTLIL representation for module `\$__ABC9_LUT6'.
Generating RTLIL representation for module `\$__ABC9_LUT7'.
Generating RTLIL representation for module `\L6MUX21'.
Generating RTLIL representation for module `\CCU2C'.
Generating RTLIL representation for module `\TRELLIS_RAM16X2'.
Generating RTLIL representation for module `\PFUMX'.
[... 998 lines cut ...]
Info: [ 80754,  80755) | 
Info: [ 80755,  80756) | 
Info: [ 80756,  80757) | 
Info: [ 80757,  80758) | 
Info: [ 80758,  80759) | 
Info: [ 80759,  80760) | 
Info: [ 80760,  80761) | 
Info: [ 80761,  80762) | 
Info: [ 80762,  80763) | 
Info: [ 80763,  80764) | 
Info: [ 80764,  80765) | 
Info: [ 80765,  80766) | 
Info: [ 80766,  80767) | 
Info: [ 80767,  80768) | 
Info: [ 80768,  80769) | 
Info: [ 80769,  80770) | 
Info: [ 80770,  80771) | 
Info: [ 80771,  80772) | 
Info: [ 80772,  80773) | 

Info: Program finished normally.
```
but I don't get any output to pack with ìcepack`.

Can someone help me?

gatecat, 2022-10-28:
You need to use `nextpnr-ecp5 --25k --package CABGA256 --json top.json --lpf top.lpf --textcfg top.config` which writes a textual to-be-packed bitstream in trellis format to `top.config`

and then use `ecppack` (_not_ `icepack`) thus `ecppack [args] top.config top.bit`

(`[args]` could include `--compress` to generate a compressed bitstream, `--svf top.svf` to generate an svf file for generic jtag programming tools, etc, or nothing)

Chandler-Kluser, 2022-10-28:
Thanks, sorry for the issue

Problem solved, you are great!! :)

closed 2022-10-28 by Chandler-Kluser