YosysHQ/nextpnr #1057: nextpnr-gowin -> ERROR: Incorrect chipdb version 1 is used. Version 2 is required

federico-borello, 2022-12-06:
I'm new to this, this is my first issue generated. 
hope it's somewhat relevant and helpful

##Error generated while 'making' apicula's blinky example. Yosys .json working well. nextpnr chipdb failure.
 
###Version:
`nextpnr-gowin --version
nextpnr-gowin -- Next Generation Place and Route (Version nextpnr-0.4-47-gdb25c5c8)
`

###Command:
`nextpnr-gowin --json blinky-tangnano9k-synth.json --write blinky-tangnano9k.json --device GW1NR-LV9QN88PC6/I5 --family GW1N-9C --cst tangnano9k.cst`

###Command Output:
`ERROR: Incorrect chipdb version 1 is used. Version 2 is required
0 warnings, 1 error
make: *** [Makefile:17: blinky-tangnano9k.json] Error 255
`

##Solved if version 0.3 of nextpnr is used

gatecat, 2022-12-07:
Have you tried updating apycula and rebuilding?

gatecat, 2022-12-07:
cc @pepijndevos @yrabbit

pepijndevos, 2022-12-07:
For now install version 0.5.1a1

I've made a new release which will be on pypi in a few hours if all goes well.

federico-borello, 2022-12-07:
Perfect, after installing latest apycula from pypi works with the latest nextpnr version.
thank you!

[image]

closed 2022-12-07 by federico-borello