YosysHQ/nextpnr #1063: Issues compiling when nextpnr (can't find python modules)

roby2014, 2022-12-16:
Hi, im trying to use nextpnr + trellis.
This is how I installed trellis: `sudo pacman -S prjtrellis`

And as you can see:
```
❯ pacman -Ql prjtrellis
prjtrellis /usr/
prjtrellis /usr/bin/
prjtrellis /usr/bin/ecpbram
prjtrellis /usr/bin/ecpmulti
prjtrellis /usr/bin/ecppack
prjtrellis /usr/bin/ecppll
prjtrellis /usr/bin/ecpunpack
prjtrellis /usr/lib/
prjtrellis /usr/lib/trellis/
prjtrellis /usr/lib/trellis/libtrellis.so
prjtrellis /usr/lib/trellis/pytrellis.so
prjtrellis /usr/share/
prjtrellis /usr/share/licenses/
prjtrellis /usr/share/licenses/prjtrellis/
prjtrellis /usr/share/licenses/prjtrellis/LICENSE
prjtrellis /usr/share/trellis/
prjtrellis /usr/share/trellis/misc/
prjtrellis /usr/share/trellis/misc/basecfgs/
prjtrellis /usr/share/trellis/misc/basecfgs/README.md
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5u-25f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5u-45f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5u-85f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5um-25f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5um-45f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5um-85f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5um5g-25f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5um5g-45f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_lfe5um5g-85f.config
prjtrellis /usr/share/trellis/misc/basecfgs/empty_machxo2-1200hc.config
prjtrellis /usr/share/trellis/misc/openocd/
prjtrellis /usr/share/trellis/misc/openocd/ecp5-evn.cfg
prjtrellis /usr/share/trellis/misc/openocd/ecp5-versa.cfg
prjtrellis /usr/share/trellis/misc/openocd/ecp5-versa5g.cfg
prjtrellis /usr/share/trellis/misc/openocd/trellisboard.cfg
prjtrellis /usr/share/trellis/misc/openocd/ulx3s.cfg
prjtrellis /usr/share/trellis/misc/openocd/ulx3s_85k.cfg
prjtrellis /usr/share/trellis/timing/
prjtrellis /usr/share/trellis/timing/util/
prjtrellis /usr/share/trellis/timing/util/.gitignore
prjtrellis /usr/share/trellis/timing/util/__init__.py
prjtrellis /usr/share/trellis/timing/util/cell_fuzzers.py
prjtrellis /usr/share/trellis/timing/util/cell_html.py
prjtrellis /usr/share/trellis/timing/util/cell_timings.py
prjtrellis /usr/share/trellis/timing/util/design_pip_classes.py
prjtrellis /usr/share/trellis/timing/util/extract_ncl_routing.py
prjtrellis /usr/share/trellis/timing/util/interconnect_html.py
prjtrellis /usr/share/trellis/timing/util/parse_sdf.py
prjtrellis /usr/share/trellis/timing/util/pip_classes.py
prjtrellis /usr/share/trellis/timing/util/timing_dbs.py
prjtrellis /usr/share/trellis/timing/util/timing_solver.py
prjtrellis /usr/share/trellis/util/
prjtrellis /usr/share/trellis/util/common/
prjtrellis /usr/share/trellis/util/common/__init__.py
prjtrellis /usr/share/trellis/util/common/database.py
prjtrellis /usr/share/trellis/util/common/devices.py
prjtrellis /usr/share/trellis/util/common/diamond.py
prjtrellis /usr/share/trellis/util/common/isptcl.py
prjtrellis /usr/share/trellis/util/common/nets/
prjtrellis /usr/share/trellis/util/common/nets/__init__.py
prjtrellis /usr/share/trellis/util/common/nets/__main__.py
prjtrellis /usr/share/trellis/util/common/nets/ecp5.py
prjtrellis /usr/share/trellis/util/common/nets/general.py
prjtrellis /usr/share/trellis/util/common/nets/machxo2.py
prjtrellis /usr/share/trellis/util/common/nets/util.py
prjtrellis /usr/share/trellis/util/common/tiles.py
``` They are at `/usr/share/trellis/`

However, when building nextpnr, I'm getting this errors:
```
❯ cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/share/trellis/
-- Building with IPO
-- Found Python3: /usr/include/python3.10 (found suitable version "3.10.8", minimum required is "3.5") found components: Development Development.Module Development.Embed
-- Found Boost: /usr/include (found version "1.80.0") found components: filesystem program_options iostreams system thread regex chrono atomic
-- Found Boost: /usr/include (found version "1.80.0") found components: program_options filesystem system
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Found Python3: /usr/bin/python3.10 (found suitable version "3.10.8", minimum required is "3.5") found components: Interpreter
-- Trellis install prefix: /usr/share/trellis/
-- Trellis library directory: /usr/local/lib
-- Trellis data directory: /usr/share/trellis//share/trellis
-- Using ECP5 chipdb: /home/roby/repos/nextpnr/ecp5/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /home/roby/repos/nextpnr
❯ make
[  3%] Built target bbasm
[  4%] Generating chipdb/chipdb-25k.bba
Traceback (most recent call last):
  File "/home/roby/repos/nextpnr/ecp5/trellis_import.py", line 22, in <module>
    import database
ModuleNotFoundError: No module named 'database'
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:77: ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [CMakeFiles/Makefile2:255: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

I did follow the documentation, but something is not working.
I can provide more info if needed.

gatecat, 2022-12-29:
Sorry for the delay in getting back on this one - can you try cmake with just `-DTRELLIS_INSTALL_PREFIX=/usr` (the "prefix" in this case is just `/usr`)

tallenintegsys, 2023-01-18:
I've installed things in $HOME ('~' if you like)
I tried...
`cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=~ -DCMAKE_INSTALL_PREFIX=~ .`
`make -j $(nproc)`
No joy (e.g. ... "ModuleNotFoundError: No module named 'database'")

My workaround...
`cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=~ -DCMAKE_INSTALL_PREFIX=~ .`
`export PYTHONPATH=~/share/trellis/timing/util:~/share/trellis/util/common`
`make -j $(nproc)`
Works! 
Adjust your path to suit your needs, again, I am installing this stuff to my $HOME directory.

tallenintegsys, 2023-01-18:
...and then I tried again I think something doesn't like the `~` grrr! ...so I changed the `~` to `$HOME`...

`cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=$HOME -DCMAKE_INSTALL_PREFIX=$HOME .`
`make -j $(nproc)`
Works!

closed 2023-09-13 by gatecat