YosysHQ/nextpnr #1072: Assertion failure: str.back() == '"' after putting a lot of blockquotes in top.v

cdwijs, 2022-12-31:
Summary: Nextpnr throws Assertion failure: str.back() == '"' (/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/lpf.cc:49)

Steps to reproduce:
1) Clone my repository from github:
$ git clone git@github.com:cdwijs/chubby75.git
$ cd chubby75/
$ git checkout colorlight-75E-V80
branch 'colorlight-75E-V80' set up to track 'origin/colorlight-75E-V80'.
Switched to a new branch 'colorlight-75E-V80'
$ cd 5a-75e/pin-scan/
$ make
<lot's of lines snipped>
2.48. Executing CHECK pass (checking for obvious problems).
Checking module top...
Found and reported 0 problems.

2.49. Executing JSON backend.

Warnings: 16 unique messages, 16 total
End of script. Logfile hash: bd208ae73d, CPU: user 0.21s system 0.01s, MEM: 22.98 MB peak
Yosys 0.24+10 (git sha1 3ebc50dee, clang 10.0.0-4ubuntu1 -fPIC -Os)
Time spent: 73% 15x read_verilog (0 sec), 5% 15x opt_clean (0 sec), ...
nextpnr-ecp5 --25k --package CABGA256 --speed 6 --json uart_tx.json --textcfg uart_tx_out.config --lpf top.lpf --freq 65
/home/cedric/oss-cad-suite/bin/nextpnr-ecp5: line 22: /home/cedric/.config/yosyshq/fonts.conf: No such file or directory
Info: constraining clock net 'clk_i' to 25.00 MHz
terminate called after throwing an instance of 'nextpnr_ecp5::assertion_failure'
  what():  Assertion failure: str.back() == '"' (/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/lpf.cc:49)
make: *** [Makefile:12: uart_tx_out.config] Aborted (core dumped)

Nextpnr throws the above assertion with the following commits for the repository:
$ git checkout 94f3e8752f14a8be79d8587c42734c6a88a9b52c && make clean && make 
$ git checkout aa3be7350b02bc8f137034799978e0da053aa1c5 && make clean && make 
$ git checkout 15c89375b1bbccdd3cf9451ac49d692d18b34db2 && make clean && make 
$ git checkout 1248d495334886915da40c586e409b4c4a1d29dd && make clean && make 

Nextpnr does not throw the assertion with the following commit: 
$ git checkout 7717f230d375bcdff42a1c569322501398d5a0b8  && make clean && make 
HEAD is now at 7717f23 enabled all LED's. This compiles and works in the FPGA
$ make clean && make
Info: Program finished normally.
ecppack --svf uart_tx.svf uart_tx_out.config uart_tx.bit

My assumptions:
Here* I see nextpnr is in the function "bool Arch::apply_lpf(std::string filename, std::istream &in)"
[*] https://github.com/YosysHQ/nextpnr/blob/master/ecp5/lpf.cc
It looks like it trips over my .lpf file, but that's weird, as this isn't changed in this** commit
[**] https://github.com/YosysHQ/nextpnr/blob/master/ecp5/lpf.cc
I guess nextPNR doesn't like large amounts of blockquotes in top.v  (/*  */)

Versions:
$ uname -a
Linux cedric 6.0.11-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 02 Dec 2022 17:25:31 +0000 x86_64 GNU/Linux
oss-cad-suite-linux-x64-20221231.tgz
Yosys 0.24+10 (git sha1 3ebc50dee, clang 10.0.0-4ubuntu1 -fPIC -Os)
$ nextpnr-ecp5 --help
/home/cedric/oss-cad-suite/bin/nextpnr-ecp5: line 22: /home/cedric/.config/yosyshq/fonts.conf: No such file or directory
nextpnr-ecp5 -- Next Generation Place and Route (Version nextpnr-0.4-84-geaf2bc8b)

cdwijs, 2022-12-31:
The full console output with the error: 
`
[cedric@cedric pin-scan]$ make
yosys -p "synth_ecp5 -json uart_tx.json" top.v

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.24+10 (git sha1 3ebc50dee, clang 10.0.0-4ubuntu1 -fPIC -Os)

-- Parsing `top.v' using frontend ` -vlog2k' --

1. Executing Verilog-2005 frontend: top.v
Parsing Verilog input from `top.v' to AST representation.
Storing AST representation for module `$abstract\uart_tx_8n1'.
Storing AST representation for module `$abstract\blink'.
Storing AST representation for module `$abstract\top'.
Successfully finished Verilog frontend.

-- Running command `synth_ecp5 -json uart_tx.json' --

2. Executing SYNTH_ECP5 pass.

2.1. Executing Verilog-2005 frontend: /home/cedric/oss-cad-suite/lib/../share/yosys/ecp5/cells_sim.v
Parsing Verilog input from `/home/cedric/oss-cad-suite/lib/../share/yosys/ecp5/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\LUT4'.
Generating RTLIL representation for module `\$__ABC9_LUT5'.
Generating RTLIL representation for module `\$__ABC9_LUT6'.
Generating RTLIL representation for module `\$__ABC9_LUT7'.
Generating RTLIL representation for module `\L6MUX21'.
Generating RTLIL representation for module `\CCU2C'.
Generating RTLIL representation for module `\TRELLIS_RAM16X2'.
Generating RTLIL representation for module `\PFUMX'.
Generating RTLIL representation for module `\TRELLIS_DPR16X4'.
Generating RTLIL representation for module `\DPR16X4C'.
Generating RTLIL representation for module `\LUT2'.
Generating RTLIL representation for module `\TRELLIS_FF'.
Generating RTLIL representation for module `\TRELLIS_IO'.
Generating RTLIL representation for module `\INV'.
Generating RTLIL representation for module `\TRELLIS_SLICE'.
Generating RTLIL representation for module `\DP16KD'.
Generating RTLIL representation for module `\FD1P3AX'.
Generating RTLIL representation for module `\FD1P3AY'.
Generating RTLIL representation for module `\FD1P3BX'.
Generating RTLIL representation for module `\FD1P3DX'.
Generating RTLIL representation for module `\FD1P3IX'.
Generating RTLIL representation for module `\FD1P3JX'.
Generating RTLIL representation for module `\FD1S3AX'.
Generating RTLIL representation for module `\FD1S3AY'.
Generating RTLIL representation for module `\FD1S3BX'.
Generating RTLIL representation for module `\FD1S3DX'.
Generating RTLIL representation for module `\FD1S3IX'.
Generating RTLIL representation for module `\FD1S3JX'.
[... 851 lines cut ...]
   Number of public wire bits:      18
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  0

2.48. Executing CHECK pass (checking for obvious problems).
Checking module top...
Found and reported 0 problems.

2.49. Executing JSON backend.

Warnings: 16 unique messages, 16 total
End of script. Logfile hash: bd208ae73d, CPU: user 0.21s system 0.01s, MEM: 22.98 MB peak
Yosys 0.24+10 (git sha1 3ebc50dee, clang 10.0.0-4ubuntu1 -fPIC -Os)
Time spent: 73% 15x read_verilog (0 sec), 5% 15x opt_clean (0 sec), ...
nextpnr-ecp5 --25k --package CABGA256 --speed 6 --json uart_tx.json --textcfg uart_tx_out.config --lpf top.lpf --freq 65
/home/cedric/oss-cad-suite/bin/nextpnr-ecp5: line 22: /home/cedric/.config/yosyshq/fonts.conf: No such file or directory
Info: constraining clock net 'clk_i' to 25.00 MHz
terminate called after throwing an instance of 'nextpnr_ecp5::assertion_failure'
  what():  Assertion failure: str.back() == '"' (/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/ecp5/lpf.cc:49)
make: *** [Makefile:12: uart_tx_out.config] Aborted (core dumped)
[cedric@cedric pin-scan]$ 

`

cdwijs, 2022-12-31:
The console output when everything goes OK:
`
[cedric@cedric pin-scan]$ git checkout 7717f230d375bcdff42a1c569322501398d5a0b8
HEAD is now at 7717f23 enabled all LED's. This compiles and works in the FPGA
[cedric@cedric pin-scan]$ make clean && make
rm -f *.svf *.bit *.config *.ys
nextpnr-ecp5 --25k --package CABGA256 --speed 6 --json uart_tx.json --textcfg uart_tx_out.config --lpf top.lpf --freq 65
/home/cedric/oss-cad-suite/bin/nextpnr-ecp5: line 22: /home/cedric/.config/yosyshq/fonts.conf: No such file or directory
Info: constraining clock net 'clk_i' to 25.00 MHz

Info: Logic utilisation before packing:
Info:     Total LUT4s:      2220/24288     9%
Info:         logic LUTs:    644/24288     2%
Info:         carry LUTs:   1576/24288     6%
Info:           RAM LUTs:      0/ 3036     0%
Info:          RAMW LUTs:      0/ 6072     0%

Info:      Total DFFs:       484/24288     1%

Info: Packing IOs..
Info: pin 'clk_i$tr_io' constrained to Bel 'X0/Y47/PIOC'.
Info: pin 'BALL_M4$tr_io' constrained to Bel 'X0/Y35/PIOC'.
Info: pin 'BALL_L5$tr_io' constrained to Bel 'X0/Y29/PIOD'.
Info: pin 'BALL_L4$tr_io' constrained to Bel 'X0/Y29/PIOC'.
Info: pin 'BALL_L13$tr_io' constrained to Bel 'X72/Y29/PIOC'.
Info: pin 'BALL_K5$tr_io' constrained to Bel 'X0/Y29/PIOB'.
Info: pin 'BALL_K4$tr_io' constrained to Bel 'X0/Y29/PIOA'.
Info: pin 'BALL_K13$tr_io' constrained to Bel 'X72/Y29/PIOA'.
Info: pin 'BALL_J4$tr_io' constrained to Bel 'X0/Y17/PIOC'.
Info: pin 'BALL_J13$tr_io' constrained to Bel 'X72/Y17/PIOC'.
Info: pin 'BALL_H5$tr_io' constrained to Bel 'X0/Y17/PIOA'.
Info: pin 'BALL_H3$tr_io' constrained to Bel 'X0/Y14/PIOD'.
Info: pin 'BALL_H15$tr_io' constrained to Bel 'X72/Y20/PIOB'.
Info: pin 'BALL_H14$tr_io' constrained to Bel 'X72/Y14/PIOD'.
Info: pin 'BALL_H13$tr_io' constrained to Bel 'X72/Y17/PIOB'.
Info: pin 'BALL_H12$tr_io' constrained to Bel 'X72/Y17/PIOA'.
Info: pin 'BALL_G5$tr_io' constrained to Bel 'X0/Y11/PIOA'.
Info: pin 'BALL_G4$tr_io' constrained to Bel 'X0/Y11/PIOB'.
Info: pin 'BALL_G2$tr_io' constrained to Bel 'X0/Y14/PIOB'.
Info: pin 'BALL_G15$tr_io' constrained to Bel 'X72/Y14/PIOB'.
Info: pin 'BALL_G14$tr_io' constrained to Bel 'X72/Y14/PIOC'.
Info: pin 'BALL_G13$tr_io' constrained to Bel 'X72/Y11/PIOB'.
Info: pin 'BALL_G1$tr_io' constrained to Bel 'X0/Y20/PIOA'.
Info: pin 'BALL_F5$tr_io' constrained to Bel 'X0/Y8/PIOD'.
Info: pin 'BALL_F3$tr_io' constrained to Bel 'X0/Y5/PIOD'.
Info: pin 'BALL_F2$tr_io' constrained to Bel 'X0/Y11/PIOC'.
Info: pin 'BALL_F16$tr_io' constrained to Bel 'X72/Y14/PIOA'.
Info: pin 'BALL_F14$tr_io' constrained to Bel 'X72/Y5/PIOD'.
Info: pin 'BALL_F13$tr_io' constrained to Bel 'X72/Y8/PIOC'.
Info: pin 'BALL_E3$tr_io' constrained to Bel 'X0/Y5/PIOC'.
Info: pin 'BALL_E16$tr_io' constrained to Bel 'X72/Y11/PIOD'.
Info: pin 'BALL_E15$tr_io' constrained to Bel 'X72/Y8/PIOB'.
Info: pin 'BALL_E1$tr_io' constrained to Bel 'X0/Y11/PIOD'.
Info: pin 'BALL_D4$tr_io' constrained to Bel 'X9/Y0/PIOB'.
Info: pin 'BALL_D3$tr_io' constrained to Bel 'X0/Y2/PIOD'.
Info: pin 'BALL_D1$tr_io' constrained to Bel 'X0/Y8/PIOA'.
Info: pin 'BALL_C4$tr_io' constrained to Bel 'X11/Y0/PIOA'.
Info: pin 'BALL_C16$tr_io' constrained to Bel 'X72/Y5/PIOA'.
Info: pin 'BALL_C15$tr_io' constrained to Bel 'X72/Y5/PIOB'.
Info: pin 'BALL_C14$tr_io' constrained to Bel 'X72/Y2/PIOC'.
Info: pin 'BALL_C1$tr_io' constrained to Bel 'X0/Y5/PIOA'.
Info: pin 'BALL_B15$tr_io' constrained to Bel 'X72/Y2/PIOB'.
Info: pin 'BALL_B1$tr_io' constrained to Bel 'X0/Y2/PIOA'.
Info: Packing constants..
Info: Packing carries...
Info: Packing LUTs...
Info: Packing LUT5-7s...
Info: Packing FFs...
Info:     358 FFs paired with LUTs.
Info: Generating derived timing constraints...
[... 232 lines cut ...]
Info:          + represents [1,8) endpoint(s)
Info: [  6804,   8408) |***+
Info: [  8408,  10012) |*+
Info: [ 10012,  11616) |***********************************************+
Info: [ 11616,  13220) |************************************************************ 
Info: [ 13220,  14824) |***+
Info: [ 14824,  16428) | 
Info: [ 16428,  18032) | 
Info: [ 18032,  19636) | 
Info: [ 19636,  21240) | 
Info: [ 21240,  22844) | 
Info: [ 22844,  24448) | 
Info: [ 24448,  26052) | 
Info: [ 26052,  27656) | 
Info: [ 27656,  29260) | 
Info: [ 29260,  30864) | 
Info: [ 30864,  32468) | 
Info: [ 32468,  34072) | 
Info: [ 34072,  35676) |****+
Info: [ 35676,  37280) |***+
Info: [ 37280,  38884) |+

Info: Program finished normally.
ecppack --svf uart_tx.svf
`

cdwijs, 2023-01-01:
It turns out I forgot a " on line 295 of top.lpf. it reads:
LOCATE	COMP	"BALL_M9"  	SITE	"M9;
It should have been:
LOCATE	COMP	"BALL_M9"  	SITE	"M9";

Is it possible to generate a message stating what line of the lpf is missing quotes?

gatecat, 2023-01-02:
Thanks for reporting! The error handling here should be improved in https://github.com/YosysHQ/nextpnr/pull/1075

closed 2023-01-18 by gatecat