YosysHQ/nextpnr #1079: nextpnr fails to build against boost 1.81

salfter, 2023-01-19:
One of my Gentoo boxes pulled in Boost 1.81, after which nextpnr was on the list for a rebuild.  The rebuild failed.  I masked ```>=dev-libs/boost-1.81``` to force it to fall back to 1.80 and ran the rebuild again.  This time, it succeeded.

I now have the same upgrade running on a different host to try to replicate the problem.  I'll attach the build log when the nextpnr build fails.

salfter, 2023-01-19:
Crap.  It built under Boost 1.81 on one host, but not on the other.  Need to see if I can replicate it on the first host.

salfter, 2023-01-19:
Still won't build under Boost 1.81 on at least one system: [build.log](https://github.com/YosysHQ/nextpnr/files/10460796/build.log)

gatecat, 2023-01-20:
Thanks for reporting! Those CMake warnings are probably harmless, and the basename deprecation still needs to be looked at, but the problem actually causing the build to fail should be fixed in #1080.

salfter, 2023-02-10:
Got a bit distracted...now trying to build a93f49e against Boost 1.81.  It'll be a bit, as LibreOffice, KiCad, and FreeCAD are among the packages that need to be rebuilt (though nextpnr gets done before any of them).

salfter, 2023-02-10:
...and it looks like it succeeded.

closed 2023-02-10 by salfter