YosysHQ/nextpnr #1082: ERROR: Unable to place cell '...', no BELs remaining to implement cell type 'TRELLIS_COMB'

roby2014, 2023-01-24:
Linux, Version:
```
❯ nextpnr-ecp5 --version
nextpnr-ecp5 -- Next Generation Place and Route (Version nextpnr-0.4-63-g16ffd02a)
```

Reproduction steps
```
git clone https://github.com/roby2014/iob-soc
cd iob-soc
make fpga-build BOARD=COLORLIGHT_5A-75E REVISION=8.0 INIT_MEM=1 RUN_EXTMEM=1
```

Actual behaviour (place & route crashes with error):
```
ERROR: Unable to place cell 'system.ext_mem0.dcache.cache_memory.n_ways_block[1].line2mem_block[4].BE_FE_block[0].cache_memory.ram[2].iob_cache_mem.ram.0.1$DPRAM_COMB0', no BELs remaining to implement cell type 'TRELLIS_COMB'
```

Expected behaviour:
Bitstream generation (it compiles with quartus/vivado toolchain)

Full log files in case it helps (synth and pnr):
https://easyupload.io/m/wk2v03

rowanG077, 2023-01-24:
The logic utilization is simply way too high. Unless there is something really weird going on the project simply won't fit on an ECP5 with 24k LUTs.

Does Diamond even succeed here? That it works for a different FPGA doesn't say much as the resource available can be orders of magnitude different.

roby2014, 2023-01-24:
It should not use so many LUTS (at least, it does not with other compilation tools, that what I meant) so I suppose its something about yosys/nextpnr that is weird.

rowanG077, 2023-01-24:
Logic utilization is not directly comparable between vendors. Especially not if your project uses inference for FPGA specific primitives. Which could lead to them being emulated for high resource cost. 

If anything this should be an issue in [yosys](https://github.com/YosysHQ/yosys) as yosys is responsible for creating the netlist. I would advice you to first try the same design with Lattice Diamond to see if this is truly a yosys issue.

closed 2023-01-25 by gatecat

gatecat, 2023-01-25:
The first thing to look at here is probably whether memory inference is going wrong somehow - although that's squarely a Yosys issue as already mentioned.