YosysHQ/nextpnr #1083: ERROR: net '...' user port 'CE' missing on cell '$PACKER_GND' possibly related to syn_useioff for ECP5

jputnam, 2023-01-25:
I got an internal check error while trying to work with SDRAM.  The error message plus a few lines of context is:

Info: Promoting globals...
Info:     promoting clock net clk_25mhz$TRELLIS_IO_IN to global network
ERROR: net 'ready' user port 'CE' missing on cell '$PACKER_GND'
ERROR: INTERNAL CHECK FAILED: please report this error with the design and full log output. Failure details are above this message.
ERROR: Packing design failed.
0 warnings, 3 errors

I've reduced the example as much as I could.  The full version actually causes nextpnr to segfault, but for now I don't have the energy to reduce that in a way that keeps the segfault.  I'm also not sure whether it's good or bad that the error remained after I changed from using the SDRAM control lines to the LED outputs.

Build commands:
yosys -p "synth_ecp5 -json checkfailed.json" checkfailed.v
nextpnr-ecp5 -r --json checkfailed.json --textcfg checkfailed.config --85k --package CABGA381 --lpf ulx3s.lpf

Versions:
$ yosys --version
Yosys 0.25+83 (git sha1 755b753e1, clang 10.0.0-4ubuntu1 -fPIC -Os)
$ nextpnr-ecp5 --version
nextpnr-ecp5 -- Next Generation Place and Route (Version nextpnr-0.5-13-gb9ed39bc)

Both were built from source today (2023-01-24).  The platform is Ubuntu on WSL, and I updated that just before rebuilding Yosys and nextpnr.  The uploaded .zip file also contains the JSON generated by Yosys.

[yosys-output.txt](https://github.com/YosysHQ/nextpnr/files/10495532/yosys-output.txt)
[checkfailed.zip](https://github.com/YosysHQ/nextpnr/files/10495556/checkfailed.zip)
[nextpnr-output.txt](https://github.com/YosysHQ/nextpnr/files/10495548/nextpnr-output.txt)

jputnam, 2023-01-25:
Here's a version that segfaults.  There's a sorting dependence; if I rename line0 to zeta, the segfault goes away, and I get "user port 'CE' missing" instead.  I've included both versions.

If it helps, the first line of dmesg is:
[    0.000000] Linux version 5.10.60.1-microsoft-standard-WSL2 (oe-user@oe-host) (x86_64-msft-linux-gcc (GCC) 9.3.0, GNU
 ld (GNU Binutils) 2.34.0.20200220) #1 SMP Wed Aug 25 23:20:18 UTC 2021

and the segfault lines in dmesg look like this:
[37504.781603] nextpnr-ecp5[22067]: segfault at 664 ip 000055f61d9abecb sp 00007ffebb4c1fa0 error 4 in nextpnr-ecp5[55f61d742000+2a3000]
[37504.781612] Code: 0f af c2 48 01 c0 48 39 f8 0f 87 80 02 00 00 44 89 e8 31 d2 f7 f7 48 63 d2 48 8d 04 91 48 bf cd cc cc cc cc cc cc cc 4c 89 da <48> 63 00 4c 29 d2 48 c1 fa 03 48 0f af d7 eb 2e 0f 1f 44 00 00 48

[nextpnr-output.txt](https://github.com/YosysHQ/nextpnr/files/10495853/nextpnr-output.txt)
[nextpnr-zeta-output.txt](https://github.com/YosysHQ/nextpnr/files/10495854/nextpnr-zeta-output.txt)
[segfault.zip](https://github.com/YosysHQ/nextpnr/files/10495855/segfault.zip)
[yosys-output.txt](https://github.com/YosysHQ/nextpnr/files/10495856/yosys-output.txt)
[yosys-zeta-output.txt](https://github.com/YosysHQ/nextpnr/files/10495857/yosys-zeta-output.txt)

gatecat, 2023-01-25:
Thank you very much for the detailed bug report and for including a testcase!

Hopefully this is fixed by https://github.com/YosysHQ/nextpnr/pull/1084, please can you give it a try on your end too?

jputnam, 2023-01-25:
The non-reduced code also works, so this looks okay to close.  Thanks for the quick response.

PR #1084 (2023-01-27): ecp5: Improve IOFF CE handling robustness
Fixes #1083
closed 2023-01-25 by gatecat