YosysHQ/nextpnr #1091: How to interact with nextpnr

quantrpeter, 2023-02-06:
Hi All
   I created a tool to convert verilog into raw netlist (which is a verilog file too). May I know how to interact with nextpnr to product bitstream file? 
thanks
Peter

Ravenslofty, 2023-02-13:
Well, nextpnr doesn't take Verilog as input, it takes JSON files instead; the kind produced by Yosys. The most obvious answer would be to run your Verilog netlist file through Yosys and include `write_json output.json` to write the netlist in the correct format.

Afterwards, it's probably easiest to consult `nextpnr-$arch --help`; you're going to need at least `nextpnr-$arch --json output.json` to read the file, but things like target device type and bitstream output very much depend on the architecture in question.

quantrpeter, 2023-02-13:
thanks @Ravenslofty , how can i have the format of the json? any schema? I think I first target nextpnr-ecp5

Ravenslofty, 2023-02-13:
[this](https://github.com/YosysHQ/yosys/blob/master/backends/json/json.cc#L339-L550) is probably as close as you're going to get, I think.

closed 2023-03-08 by Ravenslofty