YosysHQ/nextpnr #1093: Which pins work for SB_GB_IO with ice40up5k sg48?

osresearch, 2023-02-06:
[image]

According to the data sheet, there are several pins that work with the global buffers in the ice40up5k, but when I try to create an `SB_GB_IO` structure with a simple test:
```
module top(
        output led_r,
        input `PIN
);
        wire clk;
        SB_GB_IO #(
                .PIN_TYPE(6'b000000),
                .IO_STANDARD("SB_LVDS_INPUT")
        ) differential_clock_input (
                .PACKAGE_PIN(`PIN),
                .GLOBAL_BUFFER_OUTPUT(clk)
        );

        reg led_r;
        
        always @(posedge clk)
                led_r <= !led_r;
endmodule
```

the only values for `PIN` that work are `gpio_23` and `gpio_37`, neither of which are listed in that table.  Am I missing something in the documentation that defines which pins are suitable?

smunaut, 2023-02-09:
Are you sure the UP5k supports LVDS input for global buffers at all ?
Try in single ended mode.

The pins that should support it are : 

* G0/46B  pin 35
* G1/45A  pin 37 
* G3/25B  pin 20
* G6/3B   pin 44

osresearch, 2023-02-11:
[image]

I can't find any details about the ice40up5k LVDS specifically, although experimentally it works with pin 37.  I'm able to drive the global buffer from the TMDS clock and route it into the PLL to reconstruct the bit-clock of the HDMI signal.

My confusion was based on the different naming conventions -- this spreadsheet from Lattice maps the `G1/45A` to pin 37/34 pair: https://www.latticesemi.com/-/media/LatticeSemi/Documents/PinPackage/iCE/iCE40UP-5k-Pinout.ashx?document_id=51971

(This doesn't explain why nextpnr let me use pin 23 to feed the global buffer; it looks like 20/13 are the right pair based on the spreadsheet, although for some reason the upduino doesn't route that pair to the outside?)

closed 2023-02-11 by osresearch

smunaut, 2023-02-11:
Ok, just checked and actually pin 23 ( IO tile 19/31/0 ) and also pin 41 ( IO tile 19/31/0 ) should have connections to global buffers even though the lattice doc/pinout doesn't say.   Haven't actually tried them on HW though.

osresearch, 2023-02-27:
And as another hard learned lesson: pin 35 is not available for input if the PLL is used. This is mentioned in [the sysCLOCK PLL appnote](https://www.latticesemi.com/-/media/LatticeSemi/Documents/ApplicationNotes/IK2/FPGA-TN-02052-1-4-iCE40-sysCLOCK-PLL-Design-User-Guide.ashx?document_id=47778) in the "Lattice Radiant Software" section in an offhand way:

> When PLL is instantiated, the GPLL pin cannot be used as other input connection - For example, setting "rstn" in the design to pin 35; it can only drive the PLL Reference clock. However, the GPLL pin can still be set as output

smunaut, 2023-02-27:
Yes. Sams is true for the complement pin (  pin 37 ) which is also blocked for inputs if you use the second output of the PLL.

Pin 23/41 might also be blocked if you use global out on the HFOSC/LFOSC respectively too. I'm not 100% about that though.