YosysHQ/nextpnr #1095: IO flipflop packing error

jeremyherbert, 2023-02-09:
(As discussed on IRC)

I am trying to build https://github.com/joshajohnson/ecp5-mini-projects/tree/master/litex/soc-hr and I receive the following error:

```
...
Info: IOLOGIC component OFS1P3BX_1 connected to PIO Bel X38/Y0/PIOB
Info: IOLOGIC component OFS1P3BX connected to PIO Bel X35/Y0/PIOA
ERROR: CE signal or polarity mismatch for IO flipflop IFS1P3BX_1 with other IOFFs at location.
ERROR: Packing design failed.
```

Reverting this commit: https://github.com/litex-hub/valentyusb/commit/61fb144550e513be0717d3f5910a19a0cc1a6fe9 in the litex valentyusb core solves the issue. It is something to do with the allocation of IO flipflops for the USB DP/DM lines.

Please find attached the gateware folder from the failed build.

[gateware.zip](https://github.com/YosysHQ/nextpnr/files/10695837/gateware.zip)

gatecat, 2023-02-09:
Thanks, can you give https://github.com/YosysHQ/nextpnr/pull/1096 a try?

jeremyherbert, 2023-02-10:
I can confirm that the problem exists in the latest git rev, and the PR fixes the issue. Also, the resulting bit file works correctly on an FPGA (ie the USB interface works fine). Thanks for the speedy fix.

PR #1096 (2023-02-10): ecp5: Handle the case where both CE are the same constant
Fixes #1095
closed 2023-02-10 by gatecat