YosysHQ/nextpnr #1117: difference between nextpnr-fpga-interchange and nextpnr-xilinx

mohammadshahidzade, 2023-03-01:
Hi,
First of all thanks for the great project. I wanted to know the difference between nextpnr-fpga-interchange and nextpnr-xilinx and in which part of the project RapidWright is used. My guess is that RapidWright is used to generate the architecture and the place and route is done completely in nextpnr.  Also, why the fpga-xilinx run-times and frequencys are better than vivado? is it due to nextpnr or there are someparts of vivado in the project, such as vivado placement or routing?

mithro, 2023-03-01:
`nextpnr-fpga-interchange` should support any FPGA described by the FPGA interchange format.
`nextpnr-xilinx` is hard coded to support only a limited set of Xilinx Series 7 FPGAs.

Currently `nextpnr-xilinx` is significantly faster than `nextpnr-fpga-interchange` but the team at Antmicro is working hard to fix that (talk to @kgugala). The tool at https://github.com/chipsalliance/fpga-tool-perf can be used to compare the performance of different tooling and the current results can be found at https://chipsalliance.github.io/fpga-tool-perf/

RapidWright can read and write interchange formatted files, meaning that files generated from `nextpnr-fpga-interchange` can be read back into Vivado (via RapidWright) or Vivado output be pulled into `nextpnr-fpga-interchange`.

mohammadshahidzade, 2023-03-02:
Thank you very much for the response. It was really helpful. I looked at the results in https://github.com/chipsalliance/fpga-tool-perf and surprisingly the performance of nextpnr-xillinx was greater than Vivado. My guess was that the nextpnr-xillinx was using RapidWright and performing placement or routing inside Vivado to achieve these great results. However, based on your answer, I guess that nexpnr performs really well when the architecture is hard coded. can you confirm that nextpn-xilinx does not use RapidWright for placement and routing?

Ravenslofty, 2023-03-08:
nextpnr-xilinx loads the device routing and logic locations from RapidWright as part of the compile process, but doesn't access it while running. we have our own algorithms for placement (HeAP) and routing (router1/router2). these are the same algorithms that nextpnr uses for other architectures.

closed 2024-09-30 by gatecat