YosysHQ/nextpnr #1118: ice40up5k SB_RAM40_4K with different size read/write ports

osresearch, 2023-03-02:
[image]

The Lattice app note on using the embedded block rams in the ice40 mentions that there are separate controls for the read/write ports:

> The SB_RAM40_4K block has a size of 4 Kbits with separate write and read ports, each with independent control signals. By default, input and output data is 16 bits wide, although the data width is configurable using the READ_MODE and WRITE_MODE parameters

It does not say that the two modes must be the same, although it also doesn't say that they can be different... I'd like to use it as a clock cross deserializer by having a 2-bit wide write port (`.WRITE_MODE(3)`) that has an 11-bit address and a 16-bit wide read port (`.READ_MODE(0)`) with an 8-bit address.  `nextpnr-ice40` allows me to create this, although the results do not match what I would expect:

```
module dpram_2x16(
        input rd_clk,
        input [7:0] rd_addr,
        output [15:0] rd_data,
        input wr_clk,
        input wr_enable,
        input [10:0] wr_addr,
        input [1:0] wr_data,
);
        SB_RAM40_4K #(
                .WRITE_MODE(3), // x2
                .READ_MODE(0) // x16
        ) ram256x16_0 (
                .RDATA(rd_data),
                .RADDR({3'b0, rd_addr}),
                .RCLK(rd_clk),
                .RCLKE(1'b1),
                .RE(1'b1),
                .WADDR(wr_addr),
                .WCLK(wr_clk),
                .WCLKE(1'b1),
                .WE(wr_enable),
                .WDATA({14'b0, wr_data})
        );
endmodule
```

My assumption was that the 4kb would be addressed like a large linear array, but it appears that the writes are storing only 0.  Switching the code to use mode 0 (16-bit wide) for both works, so I'm fairly confident in my read/writer code.  Experimenting with the `WDATA` value, it appears that bits 11:10 and 3:2 of `WDATA` are written, but they overwrite the entire 16-bit word that is read out, so I'm quite confused about what is happening in the block ram.

gatecat, 2023-03-02:
You have to do some shuffling of the read and write data in the different modes. See the [Yosys BRAM mapping code](https://github.com/YosysHQ/yosys/blob/master/techlibs/ice40/brams_map.v).

osresearch, 2023-03-02:
Thanks, that explains things!  It looks like my experimental determination of 11 and 3 were on the right track https://github.com/YosysHQ/yosys/blob/master/techlibs/ice40/brams_map.v#L87

I'll give that mapping a try to see if that works.

osresearch, 2023-03-03:
Ok, that took way more effort than I had hoped... it seems that the address mappings in the `brams_map.v` don't work for different port sizes?  Experimentally I determined that `WADDR` for a 2-bit write port matches the shuffling, but the `RADDR` does not shuffle the bits.

What ended up working for me:

```
module dpram_2x16(
        // read domain
        input rd_clk,
        input [7:0] rd_addr,
        output [15:0] rd_data,
        // write domain
        input wr_clk,
        input wr_enable,
        input [10:0] wr_addr,
        input [1:0] wr_data,
);
        wire [10:0] RADDR = {3'b000, rd_addr[7:0]};
        wire [10:0] WADDR = {wr_addr[0], wr_addr[1], wr_addr[2], wr_addr[10:3]};

        wire [15:0] RDATA = {
                rd_data[15], rd_data[7], rd_data[11], rd_data[3],
                rd_data[13], rd_data[5], rd_data[9], rd_data[1],
                rd_data[14], rd_data[6], rd_data[10], rd_data[2],
                rd_data[12], rd_data[4], rd_data[8], rd_data[0]
        };

        wire [15:0] WDATA = { 4'b0, wr_data[1], 7'b0, wr_data[0], 3'b0 };

        SB_RAM40_4K #(
                .WRITE_MODE(3), // x2
                .READ_MODE(0) // x16
        ) ram256x16_0 (
                .RDATA(RDATA),
                .RADDR(RADDR),
                .RCLK(rd_clk),
                .RCLKE(1'b1),
                .RE(1'b1),
                .WADDR(WADDR),
                .WCLK(wr_clk),
                .WCLKE(wr_enable), // 1'b1),
                .WE(1'b1), // wr_enable),
                .WDATA(WDATA)
        );
endmodule
```

osresearch, 2023-03-04:
Ah, I think I see what's going on -- read/write address bits `[10:8]` are the two-bit-offset into the 16-bit word, so that's why they are shuffled for the 2-bit wide writes, but ignored for the 16-bit reads.

By pinning the block ram to a location next to the IO pins, I'm able to achieve over 125 MHz timings on shifting in the inputs with the DDR pins, which is much better than my clock-crossing shift register attempt and handles the HDMI input data.

closed 2023-03-04 by osresearch

smunaut, 2023-03-21:
I know you got it worked but for future reference I have this wrapper that makes things a bit more sane : https://github.com/no2fpga/no2ice40/blob/master/rtl/ice40_ebr.v