YosysHQ/nextpnr #1119: Router1 fails with index error assertion, invalid destination wire

davidlenfesty, 2023-03-04:
I have a design, using Amaranth, that fails on nextpnr routing. It succeeds with router2. It fails with the following error:

```
terminate called after throwing an instance of 'nextpnr_ecp5::assertion_failure'
  what():  Assertion failure: index < length (/work/_builds/linux-x64/nextpnr-ecp5/nextpnr/common/kernel/relptr.h:54)
build_top.sh: line 11: 165678 Aborted                 (core dumped) "$NEXTPNR_ECP5" --quiet --log top.tim --45k --package CABGA381 --speed 6 --json top.json --lpf top.lpf --textcfg top.config
```

I have replicated this with a build from oss-cad-suite from a couple months ago (unsure of the version), and the 2023-03-04 build. It also fails on a debug build from master (commit 2f509734dff676b7543a574b64a46be4224a5aa4).

I've zipped up: my debug build of nextpnr-ecp5, a coredump of the issue (using that debug build), and the build folder generated by amaranth. Too big for github, so [it's here](https://nextcloud.lenfesty.ca/s/DdDbcbtqFeBe3kF). (I can't remember if the logs are matched to the actual run but I think the bug *should* be reproducable just be re-running `build_top.sh` anyways)

davidlenfesty, 2023-03-04:
Bit of an update, turns out router2 doesn't actually work, and I've tracked down to an actual issue on my end.

Although seems like an error case router1 should ideally be able to catch.

The interaction: I'm including a generated liteeth instance into an amaranth project, which expects the MDIO to be the "raw" pin, and instantiates it's own IO buffers. On the amaranth side, it is auto-generating constraints and an IOBUF instance on the MDIO pin, and was passing the group of signals (mdio_o, mdio_i, and mdio_tristate) down into the liteeth port.

So basically it went:
MDIO FPGA pad -> IOBUF -> rgmii_mdio_i -> IOBUF, and I was getting this error from router2:

```
ERROR: No wire found for port B on destination cell eth.core.TRELLIS_IO.
```

I am assuming this is simply an error case that isn't properly handled in router1.

gatecat, 2023-03-06:
Thanks for reporting, the same error handling in router2 has been added to router1 in https://github.com/YosysHQ/nextpnr/pull/1120

closed 2023-03-10 by gatecat