YosysHQ/nextpnr #1122: Wire looks shorted in GUI (?)

trayres, 2023-03-15:
Hi everyone, 
If I select W1 in my design on the GUI, it looks like it's connected to X60/Y49/C1
[image]

If I move my mouse over slightly, it looks like the GUI is informing me that this is (in fact) connected to $PACKER_GND_NET.
[image]

W1 is correctly shown on X60/Y49/A1_SLICE however:
[image]

How can I check to make sure W1 is not shorted to $PACKER_GND_SLICE (looking through the config files?) 
I have been doing that and I *think* there isn't actually a shorting problem, but I want to confirm (...obviously).

Thank you!

gatecat, 2023-03-16:
I've checked on a known good design and this is definitely a GUI issue rather than a design issue - I have also double checked that the `Context::check()` function run in nextpnr after every pass will never allow a short circuit in the design as it ensures each wire is only used by one net.

gatecat, 2023-03-16:
See https://github.com/YosysHQ/nextpnr/pull/1123

trayres, 2023-04-04:
This appears fixed, thank you!

closed 2023-04-04 by trayres