YosysHQ/nextpnr #1127: SB_PLL40_CORE parameter DIVQ in PHASE_AND_DELAY mode

osresearch, 2023-03-18:
[image]

According to the Lattice app note, `DIVQ` is used in the `SIMPLE` feedback mode and should not affect the PLL output frequency.  Experimentally however, I see a difference on the scope if they are changed:

```
SB_PLL40_CORE #(
                .FEEDBACK_PATH("PHASE_AND_DELAY"),
                .DIVR(4'd0),            // DIVR =  0
                .DIVF(7'd4),    // DIVF =  4
                .DIVQ(3'd4),            // DIVQ =  4, but unused in phase and delay?
                .FILTER_RANGE(3'b001)   // FILTER_RANGE = 1
        ) uut (
                .LOCK(locked),
                .RESETB(~reset),
                .BYPASS(1'b0),
                .REFERENCECLK(clock_in),
                .PLLOUTCORE(clock_out)
                );
```

`DIVQ=7`:
[image]

`DIVQ=4`:
[image]

osresearch, 2023-03-18:
[image]

The `icepll -S -i 12 -o 36 -n pll_3x -m` output for a 3x non-simple feedback seems to produce a 12x multiplier instead:

```
module pll_3x(
	input  clock_in,
	output clock_out,
	output locked
	);

SB_PLL40_CORE #(
		.FEEDBACK_PATH("PHASE_AND_DELAY"),
		.DIVR(4'b0000),		// DIVR =  0
		.DIVF(7'b0000010),	// DIVF =  2
		.DIVQ(3'b100),		// DIVQ =  4
		.FILTER_RANGE(3'b001)	// FILTER_RANGE = 1
	) uut (
		.LOCK(locked),
		.RESETB(1'b1),
		.BYPASS(1'b0),
		.REFERENCECLK(clock_in),
		.PLLOUTCORE(clock_out)
		);

endmodule
```

osresearch, 2023-03-18:
In order to get a 5x PLL output that is phase locked to the reference clock, these are the parameters I had to use. This does't match how to compute the actual output frequency based on the app note and `icepll` output, so I'm confused as to what is going on in the FPGA.

```
SB_PLL40_CORE #(
                .FEEDBACK_PATH("PHASE_AND_DELAY"),
                .DIVR(4'd3), // 4 - 1
                .DIVF(7'd4), // 5 - 1
                .DIVQ(3'd4), // should not matter in non-SIMPLE
                .FILTER_RANGE(3'b001)   // FILTER_RANGE = 1
        ) uut (
                .LOCK(locked),
                .RESETB(~reset),
                .BYPASS(1'b0),
                .REFERENCECLK(clock_in),
                .PLLOUTCORE(clock_out)
                );
```

gatecat, 2023-03-19:
I believe that the documents are somewhat inaccurate here, there was some better iCE40 PLL documentation made at https://github.com/YosysHQ/icestorm/wiki/iCE40-PLL-documentation that I would trust more.

However, it is possible that icepll is not taking this into account correctly either, at least for all modes.

osresearch, 2023-03-19:
[image]

That diagram explains why `PHASE_AND_DELAY` has a 4x mulitplier -- the shiftreg output has an implicit divide by four, which has the effect of mulitplying the `DIVF` numerator.  `icepll` does not seem to take this into account?

(Also, I think my measurements with `DIVQ` were wrong -- anything that is non-zero seems to work)

osresearch, 2023-03-19:
Is there a way to guarantee the phase of the PLL output relative to the input clock? I'm seeing what I think is that it locks on an arbitrary phase offset, which means that using the PLL to shift in data based on the external reference clock does not always work.  If I use an external reset switch to the PLL, I can eventually get the 5x clock to settle "close enough" for good data.

I thought `PHASE_AND_DELAY` would do this, although it doesn't seem to.  My brute force attempt to sync the PLL output to the reference clock feels really dirty -- is there a better way to do this?
```
        reg phase_reset = 1;
        reg [20:0] error_count = 0;
        always @(posedge hdmi_clk_raw)
        begin
                if (phase_reset)
                begin
                        error_count <= error_count - 1;
                        if (error_count == 1)
                                phase_reset <= 0;
                end else
                if (hdmi_locked)
                begin
                        if (bit_clk)
                                error_count <= error_count + 1;
                        else
                        if (error_count != 0)
                                error_count <= error_count - 1;

                        if (error_count[20])
                                phase_reset <= 1;
                end
        end
```

smunaut, 2023-03-21:
`DIVQ` doesn't affect the output frequency in anything other than `SIMPLE` feedback mode, but it does still affect at which frequency the VCO runs at and the VCO needs to be in the right range for the PLL to operate.

If you use `PHASE_AND_DELAY` feedback and then use `SHIFTREG_0deg` output select, the PLL output should have a deterministic output phase vs the input clock phase.

closed 2023-09-08 by gatecat