YosysHQ/nextpnr #1131: ERROR: cell type 'TRELLIS_SLICE' is unsupported (instantiated as 'SLICE0')

trayres, 2023-04-03:
I'm now building/running from HEAD of the master branch for Yosys, prjtrellis and nextpnr.

Yosys believes TRELLIS_SLICE is still a valid techlib entry for ECP5, but nextpnr says this is an unsupported cell.

It looks like TRELLIS_SLICE is now split into TRELLIS_FF and TRELLIS_COMB (by looking at the GUI); but it looks like TRELLIS_SLICE is listed as a supported primitive here (https://github.com/YosysHQ/nextpnr/blob/581682a08e6113eb8abfaf9e690e399e350e982c/ecp5/docs/primitives.md)

Is this a build issue on my end, or is the plan to move away from TRELLIS_SLICE and to TRELLIS_COMB and TRELLIS_FF?

nextpnr-ecp5 version nextpnr-0.5-98-gb36e8a30
Yosys 0.27+22 (git sha1 53c0a6b78, clang 10.0.0-4ubuntu1 -fPIC -Os)

Ravenslofty, 2023-04-05:
TRELLIS_SLICE was removed in e42e22575f20b59634f88b5cf694efdb413ff0a0 almost exactly a year ago. 

Using separate TRELLIS_COMB and TRELLIS_FF gives significant improvements in packing designs densely. In addition, it's something of a conceptual simplification: there's no explicit packing pass anymore.

I'm not gatecat, so take my opinion with a grain of salt, but to me TRELLIS_SLICE was something of an implementation detail that was exposed for the sake of speed when the nextpnr `sa` placer took minutes to run; Yosys doesn't emit it, for example. If one wished to manually control placement and fixed routing, the public interface for this is [the BEL attribute](https://github.com/YosysHQ/nextpnr/blob/master/docs/constraints.md#absolute-placement-constraints).

trayres, 2023-04-05:
Ah, I see; that makes sense to me. 

I can control placement with the Python interface using the BEL attribute, that's working (besides the consistent crash I'm getting in the GUI right now), but how can I control routing? 

Is there an example of using the Python interface to force routing? I am struggling with that; there's a component of my design that I'd like to force routing on, and then the rest of it the regular placer/router can do its job.

gatecat, 2023-04-05:
You should be able to use the `bindWire` and `bindPip` functions to bind a routing path for a net, and hopefully with `STRENGTH_LOCKED` it won't be touched by anything else along the way.

This isn't a feature that's been heavily used, so let me know if you hit problems with it...

closed 2023-09-08 by gatecat