YosysHQ/nextpnr #1144: Segmentation Fault when trying to build on Mac

JamesTimothyMeech, 2023-04-20:
Followed the instructions to install project trellis then
`
cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local
make -j$(nproc)
sudo make install
`
but failed on "make -j$(nproc)" with this error:
`
jamesmeech@Jamess-MacBook-Pro-10 nextpnr % make -j 4                                                                                                
[  3%] Generating chipdb/chipdb-25k.bba
[  3%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
/bin/sh: line 1: 16752 Segmentation fault: 11  /opt/local/bin/python3.11 /Users/jamesmeech/Desktop/nextpnr/ecp5/trellis_import.py -L /usr/local/lib/trellis -L /usr/local/share/trellis/util/common -L /usr/local/share/trellis/timing/util -p /Users/jamesmeech/Desktop/nextpnr/ecp5/constids.inc -g /Users/jamesmeech/Desktop/nextpnr/ecp5/gfx.h 25k > chipdb/chipdb-25k.bba.new
make[2]: *** [ecp5/chipdb/chipdb-25k.bba] Error 139
make[1]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

`

rowanG077, 2023-04-20:
I expect this is a duplicate of: https://github.com/YosysHQ/prjtrellis/issues/185

JamesTimothyMeech, 2023-04-20:
Thanks for the tip. Removing libpython3.11.dylib from prjtrellis/libtrellis/CMakeFiles/pytrellis.dir/link.txt did not help for me

rowanG077, 2023-04-20:
The comment did not say but I believe you have remove it prior to rebuilding prjtrellis. Did you do this?

1. Run cmake
2. Delete the link.txt
3. Run make and install

JamesTimothyMeech, 2023-04-20:
I did as you said and I unfortunately still have the same error.

JamesTimothyMeech, 2023-04-27:
Seems to be fixed by using the newest version

closed 2023-04-27 by JamesTimothyMeech