YosysHQ/nextpnr #1147: nextpnr-gowin cannot route OSER10 to ELVDS_OBUF

marsohod4you, 2023-05-03:
I am trying to use oss-cad-suite (oss-cad-suite-linux-x64-20230502.tgz) building projects for GoWIn chip.
Issue is I cannot use OSER10 with ELVDS_OBUF (on IO pins which are not TLVDS).

Reproduce is easy:
1) git clone https://github.com/YosysHQ/apicula.git
2) cd apicula/examples/deser
3) For simplicity I modify oser10-tlvds.v and set number of channels to 1:  `define CH 1
4) Try original project with commands:
  **yosys -p "read_verilog oser10-tlvds.v; synth_gowin -json deser-oser10-tlvds-tec0117-synth.json"
  nextpnr-gowin --json deser-oser10-tlvds-tec0117-synth.json --write deser-oser10-tlvds-tec0117.json --device GW1NR-UV9QN88C6/I5 --cst oser10-tlvds-tec0117.cst
  gowin_pack -d GW1N-9 -o deser-oser10-tlvds-tec0117.fs deser-oser10-tlvds-tec0117.json**

Those above commands seem work properly.

 5) Now try to migrate other IO pins which do not support True LVDS by editing assignment file oser10-tlvds-tec0117.cst:
**//IO_LOC "q_o[0]" 29;
//IO_PORT "q_o[0]" IO_TYPE=LVCMOS33;
//IO_LOC "q_o[1]" 30;
//IO_PORT "q_o[1]" IO_TYPE=LVCMOS33;
IO_LOC "q_o[0]" 80;
IO_PORT "q_o[0]" IO_TYPE=LVCMOS33;
IO_LOC "q_o[1]" 79;
IO_PORT "q_o[1]" IO_TYPE=LVCMOS33;**

So migrate from IO pins 29,30 to 80,79.  Those 80 and 79 are not true LVDS but can work as ELVDS (emulated LVDS)
And also additionally **edit oser10-tlvds.v file replacing TLVDS_OBUF with ELVDS_OBUF**

6) Try to build with above minimal changes: yosys works properly but nextpnr-gowin crashes:

** 
nextpnr-gowin --json deser-oser10-tlvds-tec0117-synth.json --write deser-oser10-tlvds-tec0117.json --device GW1NR-UV9QN88C6/I5 --cst oser10-tlvds-tec0117.cst
/home/nick/fpga/oss-cad-suite/bin/nextpnr-gowin: line 22: /home/nick/.config/yosyshq/fonts.conf: No such file or directory
Info: Series:GW1N-9 Device:GW1NR-9 Package:QFN88 Speed:C6/I5
Info: Cell sd_out not found
Info: Cell sd_out not found
Info: Cell q_o[2] not found
Info: Cell q_o[2] not found
Info: Cell q_o[3] not found
Info: Cell q_o[3] not found

Info: Packing constants..
Info: Packing Shadow RAM..
Info: Packing GSR..
Info: Packing IOs..
Info: Packing diff IOs..
Info: Packing IO logic..
terminate called after throwing an instance of 'nextpnr_gowin::assertion_failure'
  what():  Assertion failure: q0_dst != nullptr (/work/_builds/linux-x64/nextpnr-gowin/nextpnr/gowin/pack.cc:919)
Aborted (core dumped) 
**

Is it possible to fix somehow?
GoWin EDA builds project properly but I do not like GoWin EDA and want oss-cad-suite.

mole99, 2023-05-10:
ELVDS is not supported yet, but afaik it is being worked on 🙂️

mole99, 2023-05-20:
This is what your are waiting for: https://github.com/YosysHQ/apicula/pull/177