YosysHQ/nextpnr #1158: Segfault when clocking a FF from a ring oscillator

meiniKi, 2023-05-17:
Clocking FFs from a ring oscillator leads to a segmentation fault due to an infinite recursion of `find_net_drivers`.
https://github.com/YosysHQ/nextpnr/blob/7a827b1c787edb2362fffea8f072a95e331fe976/common/kernel/timing.cc#L299

1. Is there a better way to spot infinite recursions due to circle or lasso-shaped driver networks from a ring oscillator than maintaining a trace, as I did here https://github.com/meiniKi/nextpnr/commit/e69d7ebd33cac8d3599007ab103f3628ebae0b0f ?
2. The ring oscillator may be tapped at different stages; can this lead to side effects or failed timing checks?

gatecat, 2023-05-19:
1. That commit looks good - do you want to submit it as a PR?

2. As this is only used to find simply related clocks (e.g. due to clock gating primitives), I don't expect any timing check problems here. Note that nextpnr won't automatically try and determine frequency or phase for ring oscillators, they would have to be constrained manually anyway.

meiniKi, 2023-05-21:
Great. Thanks for checking @gatecat .

closed 2023-05-22 by gatecat