YosysHQ/nextpnr #1166: [ECP5] [Linux] GUI crashes when trying to assign budget - Qt5 related?

kittennbfive, 2023-05-25:
Hi,

i am on Debian 11 (but with a custom kernel == 6.3.3). I have trouble with nextpnr crashing...

**Versions**
```
$ yosys --version
Yosys 0.29+34 (git sha1 cac1bc6fb, clang 11.0.1-2 -fPIC -Os)
$ nextpnr-ecp5 --version
"nextpnr-ecp5" -- Next Generation Place and Route (Version nextpnr-0.6-11-ge5a5de53)
$ apt list --installed libqt5core5a libqt5gui5
libqt5core5a/stable,now 5.15.2+dfsg-9 amd64 [installed,automatic]
libqt5gui5/stable,now 5.15.2+dfsg-9 amd64 [installed,automatic]
```
(compiled like 1h ago)

**What is the problem?**
[files.zip](https://github.com/YosysHQ/nextpnr/files/11568434/files.zip)
With a really basic design (.v, .json and .lpf attached, CC0) the GUI crashes when trying to assign a frequency budget.

**How to reproduce / Real behaviour**
Download the files and unpack. Run `nextpnr-ecp5 --gui --25k --package CABGA256 --speed 6`. Open .json, open .lpf inside the GUI. Then run "Design/Pack" (no problem) and "Design/Assign Budget". Leave the 50MHz default and click "OK" on the dialog. nextpnr will immediatly crash with a SIGSEGV.

**Expected behaviour**
No crash ;) If i did something wrong the program should not crash but give a clear error message.

**Console output**
```
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
If that is not possible, in Qt 5 you must at least reimplement
QCoreApplication::notify() and catch all exceptions there.

qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 1511, resource id: 6573467, major code: 40 (TranslateCoords), minor code: 0
QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?
```

**gdb backtrace**
```
#0  0x00007ffff6c56990 in QTransform::type() const () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#1  0x00007ffff6bea200 in  () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#2  0x00007ffff6c0671e in QPainter::drawPixmap(QRectF const&, QPixmap const&, QRectF const&) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#3  0x00007ffff71df155 in QCommonStyle::drawControl(QStyle::ControlElement, QStyleOption const*, QPainter*, QWidget const*) const () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007ffff7245518 in QFusionStyle::drawControl(QStyle::ControlElement, QStyleOption const*, QPainter*, QWidget const*) const () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007ffff724542a in QFusionStyle::drawControl(QStyle::ControlElement, QStyleOption const*, QPainter*, QWidget const*) const () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff72eee0e in QPushButton::paintEvent(QPaintEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff71a2fae in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff716315f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00005555558900de in nextpnr_ecp5::Application::notify(QObject*, QEvent*) (this=<optimized out>, receiver=<optimized out>, event=<optimized out>) at /home/XXX/Yosys/nextpnr/gui/application.cc:125
#10 0x00007ffff64adfca in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff719b116 in QWidgetPrivate::sendPaintEvent(QRegion const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff719b962 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff719ccb3 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff719b67c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff719ccb3 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) ()
    at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff719b67c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff717386a in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff71740ab in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff71be5df in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff716315f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00005555558900de in nextpnr_ecp5::Application::notify(QObject*, QEvent*) (this=<optimized out>, receiver=<optimized out>, event=<optimized out>) at /home/XXX/Yosys/nextpnr/gui/application.cc:125
#22 0x00007ffff64adfca in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff693e9bd in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#24 0x00007ffff691364c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#25 0x00007fffee3807da in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#26 0x00007ffff5739e6b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007ffff573a118 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007ffff573a1cf in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff650551f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#30 0x00007ffff64ac98b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff64b4c00 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00005555556ad672 in nextpnr_ecp5::CommandHandler::executeMain(std::unique_ptr<nextpnr_ecp5::Context, std::default_delete<nextpnr_ecp5::Context> >) (this=0x7fffffffdce0, ctx=std::unique_ptr<nextpnr_ecp5::Context> = {...})
    at /home/XXX/Yosys/nextpnr/common/kernel/command.cc:427
#33 0x00005555556ae8f7 in nextpnr_ecp5::CommandHandler::exec() (this=0x7fffffffdce0) at /home/XXX/Yosys/nextpnr/common/kernel/command.cc:554
#34 0x000055555567b2c2 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /home/XXX/Yosys/nextpnr/ecp5/main.cc:296
```

Thank you.

rowanG077, 2023-05-25:
Budget based API is deprecated at this point. It will very soon be removed: https://github.com/YosysHQ/nextpnr/pull/1149

If I do "Design/Pack", "Design/Place" and "Design/Route" I have no issues on the current master branch of nextpnr using the json and lpf you provided.

kittennbfive, 2023-05-26:
Uh, thanks for that information!

Regarding this bug i have, could you tell me which version of Qt you are using on which OS?

rowanG077, 2023-05-26:
I noticed as well that my file picker is very slow but everything works without issue.

My system:
OS: NixOS unstable
GPU: Intel iGPU
QT: 5.15.8

kittennbfive, 2023-05-26:
Thank you!

I did some testing. Looks like the freezing bug is related to standby/wakeup-cycle of the computer, at least i rebooted and now i can again open files... However the bug i am reporting is still present, i can trigger it with "Design/Assign Budget" OR "Design/Place" too. I noticed that in this cas i have 2 messages
```
QWidget::repaint: Recursive repaint detected
QWidget::repaint: Recursive repaint detected
```
before the SIGSEGV. I am not sure if there is a bug in nextpnr or amdgpu or Qt or ...? I guess i just leave this report open and see if somebody else reports the same problem. On the command-line nextpnr is working perfectly.