YosysHQ/nextpnr #1173: gowin: incorrect implementation for LUTs with constant 1'b0 inputs

meiniKi, 2023-06-09:
Implementing non-optimized LUTs that have constant `1'b0` inputs causes incorrect results. This only happens when instantiating a LUT manually, as in

```Verilog
(* keep *) LUT2 #(
  .INIT (4'b0001)
) i_inverter (
  .I0 ( a ),
  .I1 ( 1'b0 ),
  .F  ( c ) 
);
```
because Yosys optimizes to LUT1/LUT2/LUT3 when inferring logic. 

Using the LUT2 instantiation from above, the packer receives the configuration "0001" and replicates it to obtain a 16-bit configuration string [here](https://github.com/YosysHQ/apicula/blob/d350cb7cee68144fdef159e8104bb7652ed91fdc/apycula/gowin_pack.py#L591). Thus, the configuration for the LUT is correct, independent of the valuation of I2 and I3.

When looking at the JSON output of nextpnr-gowin, GND is assumed to be the default value.

```json
          "connections": {
            "LSR": [  ],
            "CE": [  ],
            "Q": [  ],
            "F": [ 3877689 ],
            "CLK": [  ],
            "D": [  ],
            "C": [  ],
            "B": [  ],
            "A": [ 3877684 ]
          }
```

```json
        "$PACKER_GND_NET": {
          "hide_name": 1,
          "bits": [ 3877702 ] ,
          "attributes": {
            "ROUTING": " "
          }
        }
```

LUTs in gowin FPGAs, however, seem to have `1'b1` as the default input value. Changing the `INIT` parameter to `4'b0100` implements a correct design, although `I1` is set to `1'b0`. This happens only when inputs are hard-wired to `1'b0`. LUT1/LUT2/LUT3 are correct due to the config replication (same behavior for any valuation).

If my observation is correct, I propose this fix 7997f2f40f7b8624850bdece23453b5041a4ad7a.

gatecat, 2023-06-12:
@yrabbit can you have a look at this for me?

yrabbit, 2023-06-12:
> @yrabbit can you have a look at this for me?
Very interesting! I will try to give my thoughts today.

yrabbit, 2023-06-12:
@gatecat 
A completely unused and unprogrammed LUT, that is, by default, is logically INIT='16'b1111 1111 1111 1111'. So the output F of the empty, unused, unprogrammed LUT will always be 1 regardless of the state of the inputs.

However, the inputs of the default/unused etc. LUTs are not connected to VCC/VSS, but to the outputs of neighboring LUTs. 
```
Default connections A(I0), B(I1), C(I2), D(I3)

LUT#0
F5 -> A(I0) |
F3 -> B(I1) |=> F0
F4 -> C(I2) |
F2 -> D(I3) |

LUT#1
F7 -> A(I0) |
F3 -> B(I1) |=> F1
F4 -> C(I2) |
F2 -> D(I3) |

LUT#2
F5 -> A(I0) |
F1 -> B(I1) |=> F2
F6 -> C(I2) |
F0 -> D(I3) |

LUT#3
F7 -> A(I0) |
F1 -> B(I1) |=> F3
F4 -> C(I2) |
F0 -> D(I3) |

LUT#4
F7 -> A(I0) |
F1 -> B(I1) |=> F4
F6 -> C(I2) |
F0 -> D(I3) |

LUT#5
F7 -> A(I0) |
F1 -> B(I1) |=> F5
F6 -> C(I2) |
F0 -> D(I3) |

LUT#6
F5 -> A(I0) |
F1 -> B(I1) |=> F6
F4 -> C(I2) |
F0 -> D(I3) |

LUT#7
F5 -> A(I0) |
F1 -> B(I1) |=> F7
F4 -> C(I2) |
F0 -> D(I3) |
```
The fact that unused neighboring LUTs always give out on their outputs F one could in the course of experiments give the impression that the default value of inputs is VCC, but in the real situation everything depends on the neighboring.

As a result, I think to abandon the default networks for these chips and always specify the network explicitly. 

Or at least always explicitly specify the network for LUT primitives, I did not look at what the default values of others.

meiniKi, 2023-06-13:
@yrabbit thank you for looking into this and providing clarification. This is really interesting as I haven't thought about that.

I initially considered avoiding default networks, but may it worsen congestion? Another approach might be to modify the LUT configuration so that any constant input valuation on these pins leads to the same output.

meiniKi, 2023-06-13:
Great, that's good news. I'm curious to test it (a4b7816eee8e602942b6be7f1f94a8e3d758841f).

yrabbit, 2023-06-15:
> Another approach might be to modify the LUT configuration so that any constant input valuation on these pins leads to the same output.

This option is ideal at all, by the way:) Then we can leave these inputs unconnected.

meiniKi, 2023-06-15:
I will look at it and create a PR with a proposal. I'd be glad if you could provide some feedback then. :)

meiniKi, 2023-06-20:
Resolved by #1180. Thanks for your support @yrabbit .

closed 2023-06-20 by meiniKi