YosysHQ/nextpnr #1174: nextpnr-gowin: crashes with out-of-range map indexing

Xiretza, 2023-06-10:
Using latest HEAD of apicula and nextpnr:
- YosysHQ/apicula@d350cb7cee68144fdef159e8104bb7652ed91fdc
- YosysHQ/nextpnr@5b958c4d

Running the example from the apicula repo fails:

```
$ yosys -D LEDS_NR=3 -p "read_verilog blinky.v; synth_gowin -json blinky.json"
[...]
$ $ nextpnr-gowin --json blinky.json --write pnrblinky.json --device 'GW1NR-LV9QN88PC6/I5' --cst tangnano.cst
Info: Series:GW1N-9 Device:GW1NR-9 Package:QFN88P Speed:C6/I5
Info: Cell rst not found
Info: Cell LCD_B[0] not found
Info: Cell LCD_B[1] not found
Info: Cell LCD_B[2] not found
Info: Cell LCD_B[3] not found
Info: Cell LCD_B[4] not found
Info: Cell LCD_CLK not found
Info: Cell LCD_DEN not found
Info: Cell LCD_G[0] not found
Info: Cell LCD_G[1] not found
Info: Cell LCD_G[2] not found
Info: Cell LCD_G[3] not found
Info: Cell LCD_G[4] not found
Info: Cell LCD_G[5] not found
Info: Cell LCD_HYNC not found
Info: Cell LCD_R[0] not found
Info: Cell LCD_R[1] not found
Info: Cell LCD_R[2] not found
Info: Cell LCD_R[3] not found
Info: Cell LCD_R[4] not found
Info: Cell LCD_SYNC not found

Info: Packing constants..
Info: Packing Shadow RAM..
Info: Packing GSR..
Info: Packing IOs..
Info: Packing diff IOs..
Info: Packing IO logic..
Info: Packing wide LUTs..
Info: Packing LUT5s..
Info: Packing LUT6s..
Info: Packing LUT7s..
Info: Packing LUT8s..
Info: Packing ALUs..
Info: Packing LUT-FFs..
Info: Packing non-LUT FFs..
Info: Packing PLLs..
Info: Checksum: 0xde62aa40

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
Info: Checksum: 0xde62aa40

Info: Device utilisation:
Info: 	                 VCC:     1/    1   100%
Info: 	               SLICE:    54/ 8640     0%
Info: 	                 IOB:     5/  274     1%
Info: 	              OSER16:     0/   38     0%
Info: 	              IDES16:     0/   38     0%
Info: 	             IOLOGIC:     0/  296     0%
Info: 	           MUX2_LUT5:     0/ 4320     0%
Info: 	           MUX2_LUT6:     0/ 2160     0%
Info: 	           MUX2_LUT7:     0/ 1080     0%
Info: 	           MUX2_LUT8:     0/ 1056     0%
Info: 	                 GND:     1/    1   100%
Info: 	                RAMW:     0/  270     0%
Info: 	                 GSR:     1/    1   100%
Info: 	                 OSC:     0/    1     0%
Info: 	                rPLL:     0/    2     0%

[... 88 lines cut ...]
                    pi_ = 0x555555ffd2d0}}, {px = 0x555555ffd2f0, pn = {pi_ = 0x555555ffd410}}, {px = 0x555555ffd430, pn = {pi_ = 0x555555ffd560}}, {px = 0x555555ffd580, pn = {pi_ = 0x555555ffd6b0}}, {px = 0x555555ffd6d0, pn = {pi_ = 0x555555ffd810}}, {px = 0x555555ffd830, pn = {pi_ = 0x555555ffd920}}, {
                  px = 0x555555ffda00, pn = {pi_ = 0x555555ffdb00}}, {px = 0x555555ffdb20, pn = {pi_ = 0x555555ffdc30}}, {px = 0x555555ffdcb0, pn = {pi_ = 0x555555ffdda0}}, {px = 0x555555ffde80, pn = {pi_ = 0x555555ffdf80}}, {px = 0x555555ffdfa0, pn = {pi_ = 0x555555ffe0e0}}, {px = 0x555555ffe100, pn = {
                    pi_ = 0x555555ffe280}}, {px = 0x555555ffe360, pn = {pi_ = 0x555555ffe460}}, {px = 0x555555ffe540, pn = {pi_ = 0x555555ffe640}}, {px = 0x555555ffe750, pn = {pi_ = 0x555555ffe850}}, {px = 0x555555ffe960, pn = {pi_ = 0x555555ffead0}}, {px = 0x555555ffebb0, pn = {pi_ = 0x555555ffecb0}}, {
                  px = 0x555555ffecd0, pn = {pi_ = 0x555555ffede0}}, {px = 0x555555ffeec0, pn = {pi_ = 0x555555ffefa0}}, {px = 0x555555ffefc0, pn = {pi_ = 0x555555fff0f0}}, {px = 0x555555fff110, pn = {pi_ = 0x555555fff1f0}}, {px = 0x555555fff270, pn = {pi_ = 0x555555fff350}}, {px = 0x555555fff490, pn = {
                    pi_ = 0x555555fff580}}, {px = 0x555555fff5a0, pn = {pi_ = 0x555555fff6c0}}, {px = 0x555555fff800, pn = {pi_ = 0x555555fff8d0}}, {px = 0x555555fff9b0, pn = {pi_ = 0x555555fffab0}}, {px = 0x555556000050, pn = {pi_ = 0x5555560000d0}}, {px = 0x5555560080e0, pn = {pi_ = 0x5555560081b0}}, {
                  px = 0x5555560082c0, pn = {pi_ = 0x5555560083a0}}, {px = 0x5555560083c0, pn = {pi_ = 0x5555560084b0}}, {px = 0x5555560084d0, pn = {pi_ = 0x5555560085f0}}, {px = 0x555556008610, pn = {pi_ = 0x555556008730}}}, belong_to_group = std::vector<bool> of length 63, capacity 64 = {true, true, true, true,
                true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true, true,
                true, true, true, true, true, true, true, true, true}, groups = std::vector of length 2, capacity 2 = {{px = 0x555555fb90f0, pn = {pi_ = 0x555555fffe90}}, {px = 0x5555560087b0, pn = {pi_ = 0x555556008860}}}}, pos = {m_names = std::vector of length 0, capacity 0, m_trailing = "run"}, argc = 9,
            argv = 0x7fffffffe0a8, logfile = {<std::basic_ostream<char, std::char_traits<char> >> = {<std::basic_ios<char, std::char_traits<char> >> = {<std::ios_base> = {_vptr.ios_base = 0x7ffff606fe28 <vtable for std::basic_ofstream<char, std::char_traits<char> >+64>, static boolalpha = std::_S_boolalpha,
                    static dec = std::_S_dec, static fixed = std::_S_fixed, static hex = std::_S_hex, static internal = std::_S_internal, static left = std::_S_left, static oct = std::_S_oct, static right = std::_S_right, static scientific = std::_S_scientific, static showbase = std::_S_showbase,
                    static showpoint = std::_S_showpoint, static showpos = std::_S_showpos, static skipws = std::_S_skipws, static unitbuf = std::_S_unitbuf, static uppercase = std::_S_uppercase, static adjustfield = std::_S_adjustfield, static basefield = std::_S_basefield, static floatfield = std::_S_floatfield,
                    static badbit = std::_S_badbit, static eofbit = std::_S_eofbit, static failbit = std::_S_failbit, static goodbit = std::_S_goodbit, static app = std::_S_app, static ate = std::_S_ate, static binary = std::_S_bin, static in = std::_S_in, static out = std::_S_out, static trunc = std::_S_trunc,
                    static __noreplace = std::_S_noreplace, static beg = std::_S_beg, static cur = std::_S_cur, static end = std::_S_end, _M_precision = 6, _M_width = 0, _M_flags = 4098, _M_exception = std::_S_goodbit, _M_streambuf_state = std::_S_goodbit, _M_callbacks = 0x0, _M_word_zero = {_M_pword = 0x0,
                      _M_iword = 0}, _M_local_word = {{_M_pword = 0x0, _M_iword = 0}, {_M_pword = 0x0, _M_iword = 0}, {_M_pword = 0x0, _M_iword = 0}, {_M_pword = 0x0, _M_iword = 0}, {_M_pword = 0x0, _M_iword = 0}, {_M_pword = 0x0, _M_iword = 0}, {_M_pword = 0x0, _M_iword = 0}, {_M_pword = 0x0, _M_iword = 0}},
                    _M_word_size = 8, _M_word = 0x7fffffffdea8, _M_ios_locale = {static none = 0, static ctype = 1, static numeric = 2, static collate = 4, static time = 8, static monetary = 16, static messages = 32, static all = 63, _M_impl = 0x7ffff6078d20 <(anonymous namespace)::c_locale_impl>}}, _M_tie = 0x0,
                  _M_fill = 0 '\000', _M_fill_init = false, _M_streambuf = 0x7fffffffdd78, _M_ctype = 0x7ffff6078740 <(anonymous namespace)::ctype_c>, _M_num_put = 0x7ffff60786d0 <(anonymous namespace)::num_put_c>, _M_num_get = 0x7ffff60786e0 <(anonymous namespace)::num_get_c>},
                _vptr.basic_ostream = 0x7ffff606fe00 <vtable for std::basic_ofstream<char, std::char_traits<char> >+24>}, _M_filebuf = {<std::basic_streambuf<char, std::char_traits<char> >> = {_vptr.basic_streambuf = 0x7ffff606fc48 <vtable for std::basic_filebuf<char, std::char_traits<char> >+16>, _M_in_beg = 0x0,
                  _M_in_cur = 0x0, _M_in_end = 0x0, _M_out_beg = 0x0, _M_out_cur = 0x0, _M_out_end = 0x0, _M_buf_locale = {static none = 0, static ctype = 1, static numeric = 2, static collate = 4, static time = 8, static monetary = 16, static messages = 32, static all = 63,
                    _M_impl = 0x7ffff6078d20 <(anonymous namespace)::c_locale_impl>}}, _M_lock = {__data = {__lock = 0, __count = 0, __owner = 0, __nusers = 0, __kind = 0, __spins = 0, __elision = 0, __list = {__prev = 0x0, __next = 0x0}}, __size = '\000' <repeats 39 times>, __align = 0}, _M_file = {_M_cfile = 0x0,
                  _M_cfile_created = false}, _M_mode = 0, _M_state_beg = {__count = 0, __value = {__wch = 0, __wchb = "\000\000\000"}}, _M_state_cur = {__count = 0, __value = {__wch = 0, __wchb = "\000\000\000"}}, _M_state_last = {__count = 0, __value = {__wch = 0, __wchb = "\000\000\000"}}, _M_buf = 0x0,
                _M_buf_size = 8192, _M_buf_allocated = false, _M_reading = false, _M_writing = false, _M_pback = 0 '\000', _M_pback_cur_save = 0x0, _M_pback_end_save = 0x0, _M_pback_init = false, _M_codecvt = 0x7ffff60786b0 <(anonymous namespace)::codecvt_c>, _M_ext_buf = 0x0, _M_ext_buf_size = 0, _M_ext_next = 0x0,
                _M_ext_end = 0x0}}}, <No data fields>}

```
</details>

Xiretza, 2023-06-10:
It's a regression introduced by 5b958c4d80e791d924c80bca48a1c41a8037e1d8/#1171, cc @rowanG077

rowanG077, 2023-06-11:
Yes, my apologies... I will look at in detail monday morning

One thing that immediately looks weird to me is that the gowin architecture during packing creates a cell that has timing parameters that I don't think are correct. The ALU it creates from `blinky.v` for example have have the following properties:

```
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.CLK port: PORT_IN, tmg class: TMG_CLOCK_INPUT
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.LSR port: PORT_IN, tmg class: TMG_REGISTER_INPUT
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.CE port: PORT_IN, tmg class: TMG_REGISTER_INPUT
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.Q port: PORT_OUT, tmg class: TMG_REGISTER_OUTPUT
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.F port: PORT_OUT, tmg class: TMG_COMB_OUTPUT
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.A port: PORT_IN, tmg class: TMG_REGISTER_INPUT
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.B port: PORT_IN, tmg class: TMG_REGISTER_INPUT
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.C port: PORT_IN, tmg class: TMG_REGISTER_INPUT
Info: cell type SLICE, ctr_d_ALU_SUM_2_HEAD_ALULC.D port: PORT_IN, tmg class: TMG_REGISTER_INPUT
```

From what I gather from the input. `Q` and `F` are `SUM` and `COUT` respectively. While `A`, `B`, `C`, `D` are some combinational inputs.

This might not actually be the issue but it's something I noticed while looking into it.

rowanG077, 2023-06-11:
I found the problem. We basically have this structure

`VCC.V(TMG_IGNORE) -> ALU.D(TMG_REGISTER_INPUT) -> ALU.F (TMG_COMB_OUTPUT)`

in the topological order this means `ALU.D` is unrelated to `ALU.F`. But in `setup_port_domains` while walking backwards it considers `ALU.D` to be a combinational driver of `ALU.F`. So it copies over the domain from `ALU.F` to `ALU.D`. But this is then never again copied to `VCC.V`. Which means later in `walk_backwards` it wants to update the `required` time of `VCC` for the domain of `ALU.D` but it has none.

There are various ways to fix this. The first order is that the topological order is not correct. But the way the cells are structured in gowin might be the real issue.

rowanG077, 2023-06-11:
Actually it's simply missing computational paths through registered ports. I removed them in my timing PR. I have created a fix.

Xiretza, 2023-06-11:
I don't have anything more complex than a blinky yet, but that much works fine for me as well with #1175. Thanks for the speedy fix!

closed 2023-06-12 by gatecat