YosysHQ/nextpnr #1178: Info: No candidate top level modules.

philiprbrenan, 2023-06-14:
I am trying to run ```nextpnr-gowin``` against the following verilog file:

[countUp.txt](https://github.com/YosysHQ/nextpnr/files/11740569/countUp.txt)

I get these messages:
```
Info: No candidate top level modules.
ERROR: Failed to autodetect top module, please specify using --top.
```
I tried ```--top countUp``` and got:

```
ERROR: Top module 'countUp' not found!
0 warnings, 1 error
```
Please tell me how to specify the top level module correctly?

The compile sequence I am using is:
```
perl -M"MakeWithPerl" -e"MakeWithPerl::makeWithPerl" -- --run "/home/phil/perl/cpan/ZeroEmulator/verilog/countUp/pushToGitHub.pl" (in directory: /home/phil/perl/cpan/ZeroEmulator/verilog/countUp)
perl -CSDA -w  "/home/phil/perl/cpan/ZeroEmulator/verilog/countUp/pushToGitHub.pl"
/home/phil/z/yosys/oss-cad-suite/bin/yosys -p "read_verilog /home/phil/perl/cpan/ZeroEmulator/verilog/countUp/countUp.sv"
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/
 Yosys 0.30+16 (git sha1 8b2a00102, clang 10.0.0-4ubuntu1 -fPIC -Os)
-- Running command `read_verilog /home/phil/perl/cpan/ZeroEmulator/verilog/countUp/countUp.sv' --
1. Executing Verilog-2005 frontend: /home/phil/perl/cpan/ZeroEmulator/verilog/countUp/countUp.sv
Parsing Verilog input from `/home/phil/perl/cpan/ZeroEmulator/verilog/countUp/countUp.sv' to AST representation.
Generating RTLIL representation for module `\countUp'.
/home/phil/perl/cpan/ZeroEmulator/verilog/countUp/countUp.sv:0: Warning: System task `$display' outside initial block is unsupported.
Successfully finished Verilog frontend.
Warnings: 1 unique messages, 1 total
End of script. Logfile hash: 849e83000e, CPU: user 0.00s system 0.00s, MEM: 9.11 MB peak
Yosys 0.30+16 (git sha1 8b2a00102, clang 10.0.0-4ubuntu1 -fPIC -Os)
Time spent: 100% 2x read_verilog (0 sec)
/home/phil/z/yosys/oss-cad-suite/bin/yosys -p "synth_gowin  -json /home/phil/perl/cpan/ZeroEmulator/verilog/countUp/countUp.json"
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
[... 658 lines cut ...]
Generating RTLIL representation for module `\$_DFFE_PP0P_'.
Generating RTLIL representation for module `\$_DFFE_NP0P_'.
Generating RTLIL representation for module `\$lut'.
Successfully finished Verilog frontend.
1.23.2. Continuing TECHMAP pass.
No more expansions possible.
<suppressed ~23 debug messages>
1.24. Executing OPT_LUT_INS pass (discard unused LUT inputs).
1.25. Executing SETUNDEF pass (replace undef values with defined constants).
1.26. Executing HILOMAP pass (mapping to constant drivers).
1.27. Executing AUTONAME pass.
1.28. Executing HIERARCHY pass (managing design hierarchy).
1.29. Printing statistics.
1.30. Executing CHECK pass (checking for obvious problems).
Found and reported 0 problems.
1.31. Executing JSON backend.
End of script. Logfile hash: 04a89f6a6d, CPU: user 0.21s system 0.00s, MEM: 23.34 MB peak
Yosys 0.30+16 (git sha1 8b2a00102, clang 10.0.0-4ubuntu1 -fPIC -Os)
Time spent: 41% 10x read_verilog (0 sec), 15% 2x proc_dff (0 sec), ...
/home/phil/z/yosys/oss-cad-suite/bin/nextpnr-gowin -v            --json /home/phil/perl/cpan/ZeroEmulator/verilog/countUp/countUp.json --write /home/phil/perl/cpan/ZeroEmulator/verilog/countUp/countUp.pnr --device "GW1NR-LV9QN88PC6/I5" --family GW1N-9C --cst /home/phil/perl/cpan/ZeroEmulator/verilog/countUp/tangnano9k.cst --top countUp
Info: Series:GW1N-9C Device:GW1NR-9C Package:QFN88P Speed:C6/I5
ERROR: Top module 'countUp' not found!
0 warnings, 1 error
Compilation finished successfully.
```

Ravenslofty, 2023-06-14:
Usually, one specifies the top module to Yosys instead of nextpnr; e.g. `synth_gowin -top countUp -json ...`.

philiprbrenan, 2023-06-14:
Yes that resolves the problem. Specify:

```synth_gowin -top countUp```

rather than:

```nextpnr-gowin --top countUp```

Might it be worth adding a check in ```synth_gowin``` to confirm the top level module name early on?  The suggestion from: ```nextpnr-gowin``` to use the ```--top``` keyword is less effective than intended as it leads one down the wrong path.

closed 2023-06-14 by philiprbrenan