YosysHQ/nextpnr #1186: Tang Nano 20k support

dalerodgers, 2023-07-04:
Am using lushaylabs Visual Studio plugin to call parts of oss-cad-suite.

With options in the x.lushay.json file of:

   **"board": "tangnano20k"**

I can synthesize okay but the error of:

**Finished Synthesys
Starting PnR with NextPnR
    ERROR: Invalid device GW2AR-LV18QN88C8/I7
Finished PnR**

Is there support for this device yet?

Thank you

yrabbit, 2023-07-04:
The GW2NA series chips are still supported at a very early experimental stage.

The database should presumably be suitable not only for TangPrimer20k, but also for tangnano20k.

Theoretically it takes very little change in nextpnr (actually just a change in the regular expression to parse the chip name) to select the right chip package configuration from the database.

I don't have tangnano20k yet and I can't practically test it on hardware.

yrabbit, 2023-07-04:
Ah yeah, to understand what is the experimental stage - you better not flash directly to flash - if there are inaccuracies in the processing of the CFG fuse tile, you will get a brick. 

Use SRAM.

dalerodgers, 2023-07-05:
I can't do complete PNR yet so can't flash or use SRAM.   Is there a way of unlocking the experimental stuff in nextpnr?

yrabbit, 2023-07-05:
Without compiling? No. And as I wrote, you have to make changes to support GW2AR-18.

FilipSzkandera, 2024-02-26:
Have there been any improvements since?
@yrabbit did you meant the regex in the main.c file, or should the makefile be changed also?

FilipSzkandera, 2024-02-26:
Thank you very much for the quick reply; that is awesome.
Just one last question, are these changes in the current nextpr release, or do I have to build it from source? I've checked the gowin/CmakeLists.txt and there does not appear to be a Tangnano20k reference, so I'm not sure if there is a development branch specifically for this, or something else that I am no seeing.

yrabbit, 2024-02-26:
Yes, all the necessary functionality for Tangnano20k is already in the current release of nextpnr (0.7), there are a couple of my fixes awaiting consideration (https://github.com/YosysHQ/nextpnr/pull/1290 and https://github.com/YosysHQ/nextpnr/pull/1291 ), but these only improve routing and a rare use case for IOLOGIC. Of course, you need the latest release of apicula to have chip databases.

I don’t know about yosys - I worked very closely on BRAM inferring, but whether they were included in any release... I don’t know.

FilipSzkandera, 2024-02-27:
I see now, thank you! I'm running Yosys, but I am very flexible as I am still learning all this stuff (so thank you for your patience :)). 
I'm running Lushay Code with Yosys (release 2023-02-10) and I assume it is calling the legacy NextPnR (No luck with the latest nighly release of OSS Cad Suite either). Can I ~just~ switch the binaries that are being called with the himbeachel architecture ones? Or do I need to rebuild a lot of stuff?

yrabbit, 2024-02-27:
I can’t say about global recompilation, but you can probably change the nextpnr binary call. Of course, changing the arguments - they are indicated differently - see example from apicula

https://github.com/YosysHQ/apicula/blob/f43a0b7e423ea8c946ea18f11bd2d7529f776f4e/examples/himbaechel/Makefile.himbaechel#L179-L199

FilipSzkandera, 2024-02-28:
Okay, I went through and got up Apycula running with nextpnr-himbeachel, which is awesome. I'm just testing a tang-nano4k (because I have a code for it that I know is working) tang-nano20k will come next.

But when adding some very small array to my hdl, this error popped up:
`ERROR: Unable to place cell 'vga_m.video_ram_m.video_ram_array.0.0', no BELs remaining to implement cell type 'RAM16SDP4'`

I thought that bsram was supported, or am I mistaken?

(note: this test array was not placable using Yosys in my testing either, but I thought this version should be able to support it)

yrabbit, 2024-02-28:
BSRAM IS supported, but RAM16SDP4 is not that memory - it is LUTs-based memory, and the chip used in Tangnano4k is notorious for not having exactly that memory:

[image]

yrabbit, 2024-02-28:
And now something completely different (just kidding).
Synthesis naturally does not know which chip you will then specify for routing and is not aware of the limitations of a particular board, so the job of specifying which memory to use falls on you. The simplest way is to indicate that you want BSRAM:

``` verilog
    (* ram_style="block" *)
    reg [31:0] RegisterBank [0:31];
```

FilipSzkandera, 2024-02-29:
I know this got out of the scope of this issue... But thank you very much for your patience.

`found attribute 'ram_style = block' on memory top.vga_m.video_ram_m.video_ram_array, forced mapping to block RAM
<suppressed ~53 debug messages>
ERROR: no valid mapping found for memory top.vga_m.video_ram_m.video_ram_array
make: *** [run] Error 1`

yrabbit, 2024-02-29:
Synthesis could not find a suitable mapping to implement your use of memory for the existing primitives.
There could be a billion reasons, maybe you don’t use registers for the address or for data, maybe something with the write/read signals.

Since I have little understanding of synthesis, I can hardly help.

Perhaps if your project is small and is on github and uses only yosys/nextpnr/gowin_pack and has a Makefile, I would take a look, but I definitely won’t install complex things because DragonFlyBSD is not very friendly with docker, wasm, rust and other things from the evil one :)

FilipSzkandera, 2024-03-07:
Yeah, that was just my fault. I wondered if it was a Yosys or nextpr problem since there were no obvious errors or warnings listed, but yeah, my fault. Thank you very much! I was able to flash tang nano 4k & 20k successfully.

yrabbit, 2024-03-07:
Excellent!
... and this issue will hang forever :)

closed 2024-09-26 by gatecat

dev-hack95, 2024-10-19:
Any steps to build this