YosysHQ/nextpnr #1187: Segmentation fault on nextpnr_gowin

philiprbrenan, 2023-07-15:
### Version

Yosys 0.30+16 (git sha1 8b2a00102, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Running the following command on the json file in the attached zip file produces a segmentation fault.

```
nextpnr-gowin --json fpga.json --write fpga.pnr --device "GW1NR-LV9QN88PC6/I5" --family GW1N-9C --cst tangnano9k.cst
```

[nextpnr-gowin_Segmentation_fault.zip](https://github.com/YosysHQ/yosys/files/12060061/nextpnr-gowin_Segmentation_fault.zip)

A failing execution can be seen as a github action at: https://github.com/philiprbrenan/zeroLowLevel/actions/runs/5559765369/jobs/10156203544 

### Expected Behavior

A message explaining the problem so that the segmentation fault can be avoided would be useful.

### Actual Behavior

The program produces a segmentation fault.

yrabbit, 2023-07-16:
I apologize in advance, I have very little experience with designing large systems on verilog.
You have a lot of manually instantiated I/O primitives. It looks like all ports of each module are designed this way.

```
~/var/nbug/artifact% fgrep '"IBUF"' fpga.json
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
          "type": "IBUF",
```
And the gowin part of nextpnr at once without doing anything gets networks where the driver is as you can see below the output Q of some DFF and the sink is the input I of IBUF primitive. 

```
Info: dbg-driver:91 driver:heapClock_DFF_Q type:DFF
Info:   usr:heap.clock_IBUF_I/I type:IBUF
```
I was so unprepared for this situation that it didn't even occur to me to process it.
If someone more experienced says that this is a common case, I will think how to deal with such internal inter-module I/O-BUFs.

gatecat, 2023-07-16:
These inter module I/O-bufs just shouldn't exist, normally a Yosys flow with flattening would be used (as is the default for gowin)
If there's a very specific reason why you need a Yosys flow with flattening, then there's still probably an issue in Yosys if it's putting IOBUFs on anything other than the module marked as top.
How are you running Yosys synthesis exactly?

philiprbrenan, 2023-07-17:
Earlier in the sequence I ran:

```synth_gowin -noflatten -nodffe -top fpga -json $j```

I have removed the **noflatten** and  **-nodffe** keywords and the segmentation fault has disappeared.

May I urge that perhaps ```nextpnr-gowin``` should write a helpful message explaining whether it can deal with the presence of -noflatten or not rather than raising a segmentation fault as this will save time for yosys users and maintainers in the future?

closed 2023-07-17 by philiprbrenan