YosysHQ/nextpnr #1194: Segfault/exception on combinatorial looping design

tommythorn, 2023-08-18:
The design below crashed both my old, self-built, nextpnr-ecp5 and the current version from https://yowasp.org/
(I apologize for my contorted attempt at forcing yosys to keep my ring oscillator -- improvements welcome).

I've attached the relevant files:
```
module top(input clk_25mhz,
           input [6:0] btn,
           output reg [7:0] led,
           output wifi_gpio0);

   // Tie GPIO0, keep ULX3S board from rebooting                                                                                                               
   assign wifi_gpio0    = 1;

   parameter      W = 3; // Must be odd > 2                                                                                                                    
   reg [W-1:0] lfsr = 1;
   always @(posedge btn[6])
     if (btn[5] != btn[4])
       lfsr <= 0;
     else
       lfsr <= {lfsr[W-2:0],lfsr[W-1]^lfsr[W-2]};

   wire [999:0] q;
   genvar      i;
   for (i = 0; i < W; i = i + 1)
     assign q[i] = q[(i + 1) % W] ^ lfsr[i];

   always @(posedge q[0]) led[0] <= !led[0];
   always @(posedge clk_25mhz) led[1] <= btn[1];
endmodule
```
[files.zip](https://github.com/YosysHQ/nextpnr/files/12383357/files.zip)

tommythorn, 2023-08-18:
Ha, it seems it's the `wire [999:0] q;` that is the issue.

rowanG077, 2023-08-19:
I cannot duplicate the crash on current nextpnr master branch state.

tommythorn, 2023-08-19:
Sorry, it took me a good while to sort out how to build nextpnr-ecp5, again.

Indeed, it doesn't crash with a native build.  Note, I was using a different test case that also crashed:
```
module top(input [6:0] btn,
           output wire [7:0] led);

   wire   ro1;
   wire   ro2;
   wire   ro3;

   (* keep *) LUT4 #(.INIT(16'd1)) i_inv1 (.Z(ro2), .A(ro1), .B(1'b0), .C(1'b0), .D(1'b0));
   (* keep *) LUT4 #(.INIT(16'd1)) i_inv2 (.Z(ro3), .A(ro2), .B(1'b0), .C(1'b0), .D(1'b0));
   (* keep *) LUT4 #(.INIT(16'd1)) i_inv3 (.Z(ro1), .A(ro3), .B(1'b0), .C(1'b0), .D(1'b0));

   reg    flop;

   wire dclk_o = flop;

   always @(posedge ro3) begin
      flop <= ~flop;
   end

   reg [26:0] cntr = 0;
   always @(posedge flop) cntr <= cntr + 1;
   assign led = btn[0] ? cntr[26:26-7] : cntr[26-8:26-7-8];
endmodule
```

rowanG077, 2023-08-19:
The wasm version is not maintained here. I guess you'd best ask the maintainers of that version. I'm not convinced this is an actual nextpnr issue.

whitequark, 2023-08-19:
@rowanG077 I am the maintainer of the Wasm version and I am also a YosysHQ member, so it *is* effectively maintained here.

The only way to crash the Wasm runtime, besides an explicit abort, is a write to an unallocated memory. If this is not an abort then this is a bug in nextpnr.

@tommythorn Please attach a script that reproduces this issue, or at least a nextpnr command line.

whitequark, 2023-08-19:
@tommythorn Nevermind, I reproduced this. The backtrace is:

```
wasmtime._trap.Trap: error while executing at wasm backtrace:
    0: 0x97cd3 - <unknown>!<wasm function 2312>
    1: 0x8f09a - <unknown>!<wasm function 2187>
    2: 0x8f0c7 - <unknown>!<wasm function 2188>
    3: 0xa4713 - <unknown>!<wasm function 2495>
    4: 0xed1a0 - <unknown>!<wasm function 2811>
    5: 0xa621a - <unknown>!<wasm function 2508>
    6: 0xb5c55 - <unknown>!<wasm function 2609>
    7: 0xb09cc - <unknown>!<wasm function 2590>
    8: 0x327fb5 - <unknown>!<wasm function 3959>
    9: 0x97ad5 - <unknown>!<wasm function 2291>
   10: 0x2024 - <unknown>!<wasm function 23>

Caused by:
    wasm trap: wasm `unreachable` instruction executed
```

This is an explicit abort I've referred to earlier, which can have many different causes. I'll take a look.

tommythorn, 2023-08-19:
the attached zip has a Makefile, isn’t that sufficient?Tommy

rowanG077, 2023-08-19:
@whitequark I see. I thought the WASM was an unofficial/unaffiliated I guess I was wrong. Of course if nextpnr itself is really the issue it will be fixed here.

tommythorn, 2023-08-19:
The YOWASP binaries are actually pretty awesome for me as I frequently switch between many hosts and platforms (I would love it if they supported the Artix/Kintex 7 work too).

whitequark, 2023-08-19:
> I see. I thought the WASM was an unofficial/unaffiliated project I guess I was wrong. Of course if nextpnr itself is really the issue it will be fixed here.

It is unofficial in the sense that it is not released by the YosysHQ organization. However it is maintained by someone who is a part of the Yosys project, so it is _maintained here_ even if it is _unofficial_. Which is a little confusing, but such is life!

whitequark, 2023-08-19:
> I would love it if they supported the Artix/Kintex 7 work too

I am only building upstream nextpnr so as soon as that ends up in upstream nextpnr I am happy to build it.

whitequark, 2023-08-19:
@tommythorn Okay, I've triaged this. yowasp-nextpnr-ecp5 version 0.6 crashes with a stack overflow:

<details>

```
$ yowasp-nextpnr-ecp5 --ignore-loops --json ~/x/x.json --package CABGA381  --speed 6 --85k --lpf ulx3s_v20.lpf
Warning: net 'clk_25mhz' does not exist in design, ignoring clock constraint

Info: Logic utilisation before packing:
Info:     Total LUT4s:         0/83640     0%
Info:         logic LUTs:      0/83640     0%
Info:         carry LUTs:      0/83640     0%
Info:           RAM LUTs:      0/10455     0%
Info:          RAMW LUTs:      0/20910     0%

Info:      Total DFFs:         0/83640     0%

Info: Packing IOs..
Info: Packing constants..
Info: Packing carries...
Info: Packing LUTs...
Info: Packing LUT5-7s...
Info: Packing FFs...
Info:     0 FFs paired with LUTs.
Info: Generating derived timing constraints...
Info: Promoting globals...
Info: Checksum: 0x4a82ab01

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
Info: Checksum: 0x4a82ab01

Info: Device utilisation:
Info:             TRELLIS_IO:     0/  365     0%
Info:                   DCCA:     0/   56     0%
Info:                 DP16KD:     0/  208     0%
Info:             MULT18X18D:     0/  156     0%
Info:                 ALU54B:     0/   78     0%
Info:                EHXPLLL:     0/    4     0%
Info:                EXTREFB:     0/    2     0%
Info:                   DCUA:     0/    2     0%
Info:              PCSCLKDIV:     0/    2     0%
Info:                IOLOGIC:     0/  224     0%
Info:               SIOLOGIC:     0/  141     0%
Info:                    GSR:     0/    1     0%
Info:                  JTAGG:     0/    1     0%
Info:                   OSCG:     0/    1     0%
Info:                  SEDGA:     0/    1     0%
Info:                    DTR:     0/    1     0%
Info:                USRMCLK:     0/    1     0%
Info:                CLKDIVF:     0/    4     0%
Info:              ECLKSYNCB:     0/   10     0%
Info:                DLLDELD:     0/    8     0%
Info:                 DDRDLL:     0/    4     0%
Info:                DQSBUFM:     0/   14     0%
Info:        TRELLIS_ECLKBUF:     0/    8     0%
Info:           ECLKBRIDGECS:     0/    2     0%
Info:                   DCSC:     0/    2     0%
Info:             TRELLIS_FF:     0/83640     0%
Info:           TRELLIS_COMB:     0/83640     0%
Info:           TRELLIS_RAMW:     0/10455     0%

Info: Placed 0 cells based on constraints.
Info: Creating initial analytic placement for 0 cells, random placement wirelen = 0.
Info:     at initial placer iter 0, wirelen = 0
Info:     at initial placer iter 1, wirelen = 0
Info:     at initial placer iter 2, wirelen = 0
Info:     at initial placer iter 3, wirelen = 0
Info: Running main analytical placer, max placement attempts per cell = 10000.
Info: HeAP Placer Time: 0.02s
Info:   of which solving equations: 0.00s
[... 435 lines cut ...]
Info:  0.5  0.5  Source cntr_TRELLIS_FF_Q_2.Q
Info:  0.9  1.5    Net cntr[24] (5,36) -> (4,37)
Info:                Sink led_LUT4_Z_2.B
Info:                Defined in:
Info:                  x.v:20.15-20.19
Info:  0.2  1.7  Source led_LUT4_Z_2.F
Info:  1.8  3.5    Net led[5]$TRELLIS_IO_OUT (4,37) -> (0,44)
Info:                Sink led[5]$tr_io.I
Info:                Defined in:
Info:                  x.v:2.30-2.33
Info: 0.8 ns logic, 2.7 ns routing

Info: Max frequency for clock '$glbnet$dclk_o': 309.50 MHz (PASS at 12.00 MHz)
Info: Max frequency for clock            'ro3': 894.45 MHz (PASS at 12.00 MHz)

Info: Max delay <async>                -> <async>: 5.52 ns
Info: Max delay posedge $glbnet$dclk_o -> <async>: 3.51 ns

1 warning, 0 errors

Info: Program finished normally.
```
</details>

So this can probably be closed.

gatecat, 2023-08-19:
Thinking about it it was probably fixed in https://github.com/YosysHQ/nextpnr/pull/1160.

closed 2023-08-19 by gatecat

whitequark, 2023-08-19:
@tommythorn You can install nightly builds of YoWASP tools from Test PyPI:

```
pip install --pre -i https://test.pypi.org/simple/ yowasp-nextpnr-ecp5
```

tommythorn, 2023-08-20:
Sorry to keep going @whitequark -- let me know where to direct these details.

On my MacBook Pro I got
```
...
Collecting yowasp-nextpnr-ecp5-base==0.0.post3689.dev176 (from yowasp-nextpnr-ecp5-25k==0.0.post3689.dev176->yowasp-nextpnr-ecp5)
  Using cached https://test-files.pythonhosted.org/packages/41/72/6738a44a7d0883856549abadcedcff8d0db6e7d3a59944e56b5063464b92/yowasp_nextpnr_ecp5_base-0.0.post3689.dev176-py3-none-any.whl (2.6 MB)
INFO: pip is looking at multiple versions of yowasp-nextpnr-ecp5-base to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement appdirs~=1.4 (from yowasp-nextpnr-ecp5-base) (from versions: 1.4.4)
ERROR: No matching distribution found for appdirs~=1.4
```

when running
```
$ pip3 install --pre -i https://test.pypi.org/simple/ yowasp-nextpnr-ecp5
```

I didn't have the problem earlier today on a different platform (though I have no reason to think it's platform specific).

(pip install doesn't work directly on Ubuntu; something about externally managed environments and virtual environments -- Python is a massive headache)