YosysHQ/nextpnr #1195: `std::out_of_range` exception thrown while packing LRAM

robertszczepanski, 2023-08-23:
After bumping Yosys and Nextnr in [video-converter-fpga-design](https://github.com/antmicro/sdi-mipi-video-converter-fpga-design) CI failed with an error:
```
Info: Packing LRAM...
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 64) > this->size() (which is 32)
/bin/bash: line 1: 31779 Aborted                 (core dumped) nextpnr-nexus --placer-heap-timingweight 35 -l /home/runner/work/sdi-mipi-video-converter-fpga-design/sdi-mipi-video-converter-fpga-design/build/1080p_3g-2lanes/1080p_3g-2lanes_nextpnr.log --device LIFCL-40-9BG256C --pdc /home/runner/work/sdi-mipi-video-converter-fpga-design/sdi-mipi-video-converter-fpga-design/constraints/video_converter_3g-2lanes.pdc --json 1080p_3g-2lanes.json --fasm 1080p_3g-2lanes.fasm
```
After bisecting Yosys the commit https://github.com/YosysHQ/yosys/commit/9f8e039a4b10e936457fd75d589eb7ec51f0292b seems to be the first one to cause an assertion issue.

You should be able to reproduce an issue while following the [CI flow](https://github.com/antmicro/sdi-mipi-video-converter-fpga-design/blob/main/.github/workflows/build.yml) but please notice that the nextpnr repository is cloned from a fork [antmicro/nextpnr](https://github.com/antmicro/nextpnr) due to https://github.com/gatecat/prjoxide/issues/42 issue.

CI job that presents an issue: https://github.com/antmicro/sdi-mipi-video-converter-fpga-design/actions/runs/5937443968/job/16099881713

gatecat, 2023-08-23:
Can you provide the JSON, PDC, etc, everything that's needed to reproduce this on the nextpnr side?

gatecat, 2023-08-23:
> That seems odd that it would cause this issue since it doesn't affect any synthesizable code. Are you sure the issue is deterministically reproduced?

We use SV format operators in the techmap rules for nexus, because some of the cells need numeric values in string parameters to match the vendor toolchain behaviour.
nextpnr shouldn't be failing in this way regardless, but it's much easier if I have something more contained to determine if the actual bug is in yosys or nextpnr, or perhaps both.

gatecat, 2023-08-23:
Okay, the problem was simply that the Yosys change resulted in fewer leading zeros than before sometimes and nextpnr couldn't cope with that. That's now fixed in https://github.com/YosysHQ/nextpnr/pull/1196

However, for what I believe are unrelated issues the design is failing to route a the DPHY clock signals. Is this a known issue or should I look into this?

robertszczepanski, 2023-08-23:
Yes, it does indeed fail and I introduced a fix in a separate branch of our forked repository https://github.com/antmicro/nextpnr/commit/ebe11279e23a3b28abf82db85f1f0faed34d863f.

As I mentioned earlier I think it's an issue related to prjoxide: https://github.com/gatecat/prjoxide/issues/42 but I would appreciate your opinion about this.

gatecat, 2023-08-23:
The problem isn't that the clock is entering the FPGA on a dedicated pin, but rather internal to the fabric of the FPGA, some of the DPHY hard IP clock ports are not directly connected to the fabric global clock tree but require a hop through generally routing.

Your fix is correct, but I will make a pull request shortly that adds all the relevant pins of the DPHY IP as there are a couple of others that might be relevant as well.

gatecat, 2023-08-23:
I've added a relevant fix to the previous linked PR fixing the `out_of_range`.

robertszczepanski, 2023-08-23:
This seems to fix the assertion problem but there is another one which I actually encountered while bisecting Yosys before.

On latest nextpnr I am unable to route this design. It achieves thousands of iterations and doesn't seem to improve. Earlier I was able to fix this issue by changing a value of the `--placer-heap-timingweight` parameter but it's not helpful anymore.

It looks like it broke after merging https://github.com/YosysHQ/nextpnr/pull/1183 but I don't know which commit exactly.

gatecat, 2023-08-23:
Ugh, this is a bit of a long running issue with router2 and nexus. It has essentially always been unreliable and small changes can cause this to come back. Bisecting unfortunately doesn't help because invariably the found commit is some unrelated innocent change that just happens to peturbbthe particular design into a state where the router oscillates.

I'll see what I can do here to fix it for this design.

gatecat, 2023-08-24:
The routing issue should now be hopefully fixed in https://github.com/YosysHQ/nextpnr/pull/1197

robertszczepanski, 2023-08-24:
I've tested this locally and it fixes an issue, thanks!

closed 2023-08-24 by robertszczepanski