YosysHQ/nextpnr #1203: Suggested command `nextpnr-himbaechel --uarch gowin --chipdb .../himbaechel/uarch/gowin/chipdb/chipdb-GW1N-1.bin  --vopt partno=GW1NR-LV9QN88PC6/I5` fails

whitequark, 2023-08-31:
If I run `nextpnr-himbaechel --uarch gowin --chipdb .../himbaechel/uarch/gowin/chipdb/chipdb-GW1N-1.bin` it proposes adding `--vopt partno=GW1NR-LV9QN88PC6/I5`. If I do that it complains that the part does not exist.

yrabbit, 2023-08-31:
That's my fault.
You indicate the partnumber for the GW1NR-9 series because it was suggested, but you have a chip from another GW1N-1 series, which, of course, does not have such a partnumber.

Apparently we need more intellectual guidance.

gatecat, 2023-09-01:
Once I next have a moment to work on this, I will set up a mechanism so that the uarch can provide the relative chipdb path given a device name. Then this problem will disappear anyway.

yrabbit, 2023-09-01:
Oookay. 
Although, to be honest, I have little idea how this will help in this case. Well, unless you identify the device by its partnumber, which you obviously won't do ;)

gatecat, 2023-09-01:
That will be the job of the gowin-uarch

yrabbit, 2023-09-01:
I knew it:)

Ravenslofty, 2023-09-01:
Bikeshedding, but nextpnr seems to be converging on "device" as the phrase to use in this context, rather than "partno"; e.g. [Mistral](https://github.com/YosysHQ/nextpnr/blob/master/mistral/main.cc#L50), [MachXO2](https://github.com/YosysHQ/nextpnr/blob/master/machxo2/main.cc#L51), [Nexus](https://github.com/YosysHQ/nextpnr/blob/master/nexus/main.cc#L50).

gatecat, 2023-09-02:
Okay, that's ridiculous but it will need a but more thought, probably another argument then.

gatecat, 2023-09-02:
Thinking about this some more, how do you even know if you have a GW2A-18 GW2A-LV18QN88C8/I7  or a GW2AR-18C GW2A-LV18QN88C8/I7 ?

yrabbit, 2023-09-02:
you mean how do I do this? I'm looking at the electrical circuit diagram.
And I have to be careful because the commercial name of the board does not say anything: Tangnano1k is with both the GW1N-1 series chip and the GW1NZ-1 series chip. Completely different inside.

[image]

whitequark, 2023-09-02:
> you mean how do I do this? I'm looking at the electrical circuit diagram.

So the electrical circuit diagram has an unambiguous part number that does not require any other information?

yrabbit, 2023-09-02:
A! Sorry, I didn't quite understand the question. Then I look at the chip. this “C” after 2317 means that I have a soldered GW2AR-18C series chip, not a GW2AR-18 :)
[image]

gatecat, 2023-09-15:
The approach has been reworked in #1214, so the user only needs to give a `--device` argument (exceptionally a `--vopt family=` in the gowin cases where this is ambiguous); and himbaechel will determine the uarch and find the chipdb from that.

closed 2023-09-15 by gatecat