YosysHQ/nextpnr #1205: "ERROR: INTERNAL CHECK FAILED" with newer oss-cad-suite-build releases

vlang-intona, 2023-09-01:
### Version

Yosys 0.32+74 (git sha1 b739213d9, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

We're seeing errors output by `nextpnr-ecp5`. A yosys change seems to have triggered this, we suspect 52c8c28d2cd05145d5b224c56742c63e79af1e04. The last version that worked was 20230525. If I use a newer `nextpnr-ecp5` build with output generated by an older `yosys` build, it passes, so I think the problem is with `yosys`.

I made a reproduction by removing random parts of our design. Unfortunately it's still 20kloc (includes output from a well known SoC generator). The design makes no sense anymore, but it passes with older builds and fails with newer ones, so it should be suitable to analyze the problem. I intend to attach the files or post a link after I submit this issue.

This is probably a really simple mistake on our side, but I can't figure out what's wrong, and I don't know what the error is supposed to be about.

### Expected Behavior

Last line of output with  oss-cad-suite-linux-x64-20230228:
```
Info: Program finished normally.
```

### Actual Behavior

Last 4 lines of output with  oss-cad-suite-linux-x64-20230831:
```
ERROR: net 'cpu_inst.Briey8127_1_inst.axi_sdramCtrl.unburstify_buffer_transaction_write_TRELLIS_FF_Q_M' user port 'cpu_inst.Briey8127_1_inst.axi_sdramCtrl.unburstify_buffer_transaction_write_TRELLIS_FF_Q.M' connected to incorrect net 'cpu_inst.Briey8127_1_inst.axi_sdramCtrl_io_axi_arbiter_io_output_arw_halfPipe_payload_write'
ERROR: INTERNAL CHECK FAILED: please report this error with the design and full log output. Failure details are above this message.
ERROR: Packing design failed.
0 warnings, 1184 errors
```

gatecat, 2023-09-01:
Thanks for reporting this, it should be fixed in https://github.com/YosysHQ/nextpnr/pull/1206

vlang-intona, 2023-09-01:
That was extremely fast!

PR #1206 (2023-09-02): ecp5: Improve packer robustness to FF dangling M input
Fixes #1205
closed 2023-09-02 by gatecat