YosysHQ/nextpnr #1207: nexpnr-gowin crashes

Raleksan, 2023-09-04:
I am trying to use oss-cad-suite (oss-cad-suite-linux-x64-20230904.tgz) for building project for Gowin chip.
Issue is following: when I use nextpnr-gowin get such error 

```
Info: No Fmax available; no interior timing paths found in design.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 18446744073709551615) >= this->size() (which is 0)
```

Reproduce is easy:
1. Download archive and unzip it [nextpnr_report.tar.gz](https://github.com/YosysHQ/nextpnr/files/12511969/nextpnr_report.tar.gz)
2. Start "syth_for_gowin.sh"

Is it possible to fix it somehow?
Gowin EDA builds project properly but I want to support Yosys-based toolchain too.

yrabbit, 2023-09-04:
it all crashes on line 90
https://github.com/YosysHQ/nextpnr/blob/79c6840fefcf84d001b0c1b269cd0198ee9d3dba/json/jsonwrite.cc#L90

and it collapses because a port with the name SW[-1] arrives to it, this -1 leads to the construction of a vector of size 0 and the size does not increase, because it calmly passes the check on line 88 (-1 is not greater than 0).

Now about SW[-1] - this is ctx->ports, but I never touch ports and I don't care what they have in their names and I don't change their names.

But in the input JSON after yosys it looks like:

```
    "board_specific_top": {
      "attributes": {
        "dynports": "00000000000000000000000000000001",
        "top": "00000000000000000000000000000001",
        "src": "board_specific_top.sv:3.1-152.10"
      },
      "parameter_default_values": {
        "clk_mhz": "00000000000000000000000000011011",
        "w_digit": "00000000000000000000000000000000",
        "w_gpio": "00000000000000000000000000000011",
        "w_key": "00000000000000000000000000000010",
        "w_led": "00000000000000000000000000000110",
        "w_sw": "00000000000000000000000000000000"
      },
      "ports": {
        "CLK": {
          "direction": "input",
          "bits": [ 2 ]
        },
        "KEY": {
          "direction": "input",
          "bits": [ 3, 4 ]
        },
        "SW": {
          "direction": "input",
          "offset": -1,
          "upto": 1,
          "bits": [ 5, 6 ]
        },
        "UART_RX": {
          "direction": "input",
          "bits": [ 7 ]
        },
```

If the elders of nextpnr tell me that I should do something with these "offset" and "upto", then of course I will. But I repeat - I don’t care what is written in the port names, for me they are just names.
I could be terribly wrong

gatecat, 2023-09-04:
Yes, it looks like this is a generic bug with negative indices in the JSON backend rather than anything gowin specific, I will look into it.

gatecat, 2023-09-04:
This should hopefully be fixed in https://github.com/YosysHQ/nextpnr/pull/1209

Raleksan, 2023-09-06:
I build nextpnr-gowin based on your commit and project is working now.
Thanks for your help!

P.S. Should I close this issue? Or it will be closed automatically after merging of pull-request?

yrabbit, 2023-09-06:
> P.S. Should I close this issue? Or it will be closed automatically after merging of pull-request? 
Close it, please ;)

closed 2023-09-06 by Raleksan