YosysHQ/nextpnr #1211: Hang in nextpnr-ice40 sa placer for SmolDVI design

MichaelBell, 2023-09-06:
Building the example design for Icebrekaer from this repo https://github.com/Wren6991/SmolDVI, but using `-abc 9 -device u` as options to yosys, I see nextpnr hang on maybe 20% of runs.

This is a repro case:
nextpnr-ice40 --placer sa --up5k --package sg48 --pcf smoldvi_icebreaker.pcf --json smoldvi_icebreaker.json  --asc smoldvi_icebreaker.asc --pre-pack smoldvi_icebreaker_prepack.py --timing-allow-fail --quiet --log pnr.log --seed 715589835
[repro.zip](https://github.com/YosysHQ/nextpnr/files/12541491/repro.zip)

gatecat, 2023-09-07:
I'll look into this, but in a meantime is removing `--placer sa` and using the default HeAP placer a workaround for now? There should be relatively few cases where SA is actually giving better performance now, and some of those should perhaps really be dealt with by HeAP improvements.

MichaelBell, 2023-09-07:
SA does seem to help a little, this is trying to meet timing for HDMI (and failing, but only fractionally with a good seed).