YosysHQ/nextpnr #1221: Trellis ECP5 python module issue

wdevore, 2023-09-28:
I just performed a git-pull and then attempted to make the project. It now errors out. Note I pulled and built Trellis first.

```
[  1%] Generating chipdb/chipdb-25k.bba
[  4%] Built target bbasm
Traceback (most recent call last):
  File "/home/xxx/Downloads/nextpnr/ecp5/trellis_import.py", line 709, in <module>
    main()
  File "/home/xxx/Downloads/nextpnr/ecp5/trellis_import.py", line 692, in main
    chip = pytrellis.Chip(dev_names[args.device])
RuntimeError: No such node (row_bias)
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:65: ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [CMakeFiles/Makefile2:259: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
```
Cmake built the makefile without zero issues:
```
nextpnr$ cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local
-- Building with IPO
-- Found Python3: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable version "3.8", minimum required is "3.5") found components: Development 
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/include (found version "1.71.0") found components: program_options filesystem system 
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Found Python3: /home/xxx/.espressif/python_env/idf4.3_py3.8_env/bin/python3.8 (found suitable version "3.8.10", minimum required is "3.5") found components: Interpreter 
-- Trellis install prefix: /usr/local
-- Trellis library directory: /usr/local/lib/trellis
-- Trellis data directory: /usr/local/share/trellis
-- Using ECP5 chipdb: /home/xxx/Downloads/nextpnr/ecp5/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xxx/Downloads/nextpnr
```
Any ideas?
Thanks.

whitequark, 2023-09-28:
You probably need a `git submodule update`

mmicko, 2023-09-28:
@wdevore for prjtrellis you also need to install, and  as @whitequark  mentioned make sure database submodule in prjtrellis is up to date.

wdevore, 2023-09-28:
Yup that fixed it it. Thanks @whitequark and @mmicko :-)

closed 2023-09-28 by wdevore