YosysHQ/nextpnr #1229: Build Issues on Windows

alchitry, 2023-10-26:
I'm trying to build this on Windows and have run into a few issues including a linking issue I can't figure out.

I'm running cmake like this.
```
cmake . -DARCH=ice40 -DICESTORM_INSTALL_PREFIX=C:\Users\justi\OneDrive\Documents\icestorm -DCMAKE_TOOLCHAIN_FILE=C:/vcpkg/scripts/buildsystems/vcpkg.cmake -DVCPKG_TARGET_TRIPLET=x64-windows-static -DSTATIC_BUILD=ON -G "Visual Studio 17 2022" -DPython3_EXECUTABLE=C:/python-3.11/python.exe -DPython3_LIBRARY=C:/vcpkg/packages/python3_x64-windows-static/lib/python311.lib -DPython3_INCLUDE_DIR=C:/vcpkg/packages/python3_x64-windows-static/include/python3.11 .
```
and building like this
```
cmake --build . --config Release
```

I've also tried a non-static build and I have the same problems.

Here's the output from the first command.
```
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.19045.
-- The CXX compiler identification is MSVC 19.37.32825.0
-- The C compiler identification is MSVC 19.37.32825.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.37.32822/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building with IPO
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE
-- Found Python3: C:/python-3.11/python.exe (found suitable version "3.11.5", minimum required is "3.5") found components: Interpreter
-- Found Python3: optimized;C:/vcpkg/packages/python3_x64-windows-static/lib/python311.lib;debug;C:/vcpkg/installed/x64-windows-static/debug/lib/python311_d.lib (found suitable version "3.11.5", minimum required is "3.5") found components: Development Development.Module Development.Embed
-- Found ZLIB: optimized;C:/vcpkg/installed/x64-windows-static/lib/zlib.lib;debug;C:/vcpkg/installed/x64-windows-static/debug/lib/zlibd.lib (found version "1.3.0")
-- Found Boost: C:/vcpkg/installed/x64-windows-static/include (found version "1.83.0") found components: filesystem program_options iostreams system thread regex chrono atomic
-- Found Boost: C:/vcpkg/installed/x64-windows-static/include (found version "1.83.0") found components: program_options filesystem system
-- Configuring architecture: ice40
-- Enabled iCE40 devices: 384;1k;5k;u4k;8k
-- Found Python3: C:/python-3.11/python.exe (found suitable version "3.11.5", minimum required is "3.5") found components: Interpreter
-- IceStorm install prefix: C:\Users\justi\OneDrive\Documents\icestorm
-- icebox data directory: C:\Users\justi\OneDrive\Documents\icestorm/share/icebox
-- Using iCE40 chipdb: C:/Users/justi/nextpnr2/ice40/chipdb
-- Configuring done (13.8s)
-- Generating done (0.0s)
-- Build files have been written to: C:/Users/justi/nextpnr2
```

When building, I run into a couple of syntax errors.
```
C:\Users\justi\nextpnr2\common\kernel\command.cc(136,10): error C2664: 'DWORD GetModuleFileNameA(HMODULE,LPSTR,DWORD)':
 cannot convert argument 2 from 'WCHAR [261]' to 'LPSTR' [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
C:\Users\justi\nextpnr2\common\kernel\command.cc(136,31): message : Types pointed to are unrelated; conversion requires
 reinterpret_cast, C-style cast or parenthesized function-style cast [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\um\libloaderapi.h(208,1): message : see declaration of 'Get
ModuleFileNameA' [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
C:\Users\justi\nextpnr2\common\kernel\command.cc(136,10): message : while trying to match the argument list '(int, WCHA
R [261], int)' [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
C:\Users\justi\nextpnr2\common\kernel\command.cc(138,10): error C2664: 'DWORD GetShortPathNameA(LPCSTR,LPSTR,DWORD)': c
annot convert argument 1 from 'WCHAR [261]' to 'LPCSTR' [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
C:\Users\justi\nextpnr2\common\kernel\command.cc(138,27): message : Types pointed to are unrelated; conversion requires
 reinterpret_cast, C-style cast or parenthesized function-style cast [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
C:\Program Files (x86)\Windows Kits\10\Include\10.0.22621.0\um\winbase.h(1375,1): message : see declaration of 'GetShor
tPathNameA' [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
C:\Users\justi\nextpnr2\common\kernel\command.cc(138,10): message : while trying to match the argument list '(WCHAR [26
1], TCHAR [261], int)' [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
[... 171 lines cut ...]
    Hint on symbols that are defined and could potentially match:
      "void __cdecl nextpnr_ice40::rdft(int,int,float *,int *,float *)" (?rdft@nextpnr_ice40@@YAXHHPEAMPEAH0@Z)
fftsg2d.obj : error LNK2001: unresolved external symbol "void __cdecl cdft(int,int,float *,int *,float *)" (?cdft@@YAXHHPEAMPEAH0@Z) [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
    Hint on symbols that are defined and could potentially match:
      "void __cdecl nextpnr_ice40::cdft(int,int,float *,int *,float *)" (?cdft@nextpnr_ice40@@YAXHHPEAMPEAH0@Z)
fftsg2d.obj : error LNK2001: unresolved external symbol "void __cdecl cdft2d_sub(int,int,int,float * *,float *,int *,float *)" (?cdft2d_sub@@YAXHHHPEAPEAMPEAMPEAH1@Z) [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
    Hint on symbols that are defined and could potentially match:
      "void __cdecl nextpnr_ice40::cdft2d_sub(int,int,int,float * *,float *,int *,float *)" (?cdft2d_sub@nextpnr_ice40@@YAXHHHPEAPEAMPEAMPEAH1@Z)
C:\Users\justi\nextpnr2\Release\nextpnr-ice40.exe : fatal error LNK1120: 37 unresolved externals [C:\Users\justi\nextpnr2\nextpnr-ice40.vcxproj]
```

This seems to be a problem with the fftsg files.

Digging into them it looks like a bunch of the functions declare function prototypes in them.
```
void cdft(int n, int isgn, float* a, int* ip, float* w)
{
  void makewt(int nw, int* ip, float* w);
  void cftfsub(int n, float* a, int* ip, int nw, float* w);
  void cftbsub(int n, float* a, int* ip, int nw, float* w);
  int nw;
```
These are defined later in the same file (and in the same `NEXTPNR_NAMESPACE_BEGIN` `NEXTPNR_NAMESPACE_END` block) so I'm not sure why it is failing to link these. It seems like the prototypes aren't being marked as in the namespace.

I'm not sure how to fix this and any help is greatly appreciated.

alchitry, 2023-10-26:
Ok well I found another work around.

In `nextpnr_namespaces.h` I changed
```
#ifdef NEXTPNR_NAMESPACE
#define NEXTPNR_NAMESPACE_PREFIX NEXTPNR_NAMESPACE::
#define NEXTPNR_NAMESPACE_BEGIN namespace NEXTPNR_NAMESPACE {
#define NEXTPNR_NAMESPACE_END }
#define USING_NEXTPNR_NAMESPACE using namespace NEXTPNR_NAMESPACE;
#else
#define NEXTPNR_NAMESPACE_PREFIX
#define NEXTPNR_NAMESPACE_BEGIN
#define NEXTPNR_NAMESPACE_END
#define USING_NEXTPNR_NAMESPACE
#endif
```
to simply
```
#define NEXTPNR_NAMESPACE_PREFIX
#define NEXTPNR_NAMESPACE_BEGIN
#define NEXTPNR_NAMESPACE_END
#define USING_NEXTPNR_NAMESPACE
```

It now compiled fine. I don't know what side effects removing the namespace could have but it fully builds now.

whitequark, 2023-10-26:
> If I change `__MINGW32__` to `_WIN32` then it resolves it (`longpath` and `shortpath` are then `char`).

This is the correct fix. Could you submit a PR please?

> The rest of the errors seem to be a macro expansion problem. `std::to_string(std::max(0, vm["placer-heap-cell-placement-timeout"].as<int>()));`
> 
> I found [this](https://stackoverflow.com/questions/2789481/problem-calling-stdmax) which suggested wrapping `std::max` in () and that works.

This works but I would rather see `#define NOMINMAX` so that `std::max` can be used freely.

> Digging into them it looks like a bunch of the functions declare function prototypes in them.

The prototypes should probably be defined at the beginning of the file instead of this.

alchitry, 2023-10-26:
I created a PR (#1230) with all the fixes.

PR #1230 (2023-10-26): Windows fixes
Closes #1229 

Adds `#NOMINMAX` to all includes of `windows.h` to prevent the min/max macros from causing problems.

Replaces `__MINGW32__` with `_WIN32` in `command.cc`

Moves prototypes out of functions and to the top of `fftsg.cc` and `fftsg2d.cc` to prevent linker errors.
closed 2023-10-26 by gatecat