YosysHQ/nextpnr #1231: arachne-pnr or nextpnr for ICE40 devices

PythonLinks, 2023-10-28:
Thank you for the excellent review of the different software packages.  I was not even aware of some of them. 

Since I am only using an ICE40 device, it is not clear to me if I am better served by the arachen-pnr which is icestorm specific, or is it better to use your nextprn-ice40.   

Maybe what I should do is an experiment, to see which generates a smaller faster device. 

Could you be so kind as to update the docs with your recommendation? 

And thank you for all that you have done so far.

whitequark, 2023-10-28:
Arachne is superseded by nextpnr.

Ravenslofty, 2023-10-29:
At the very top of the readme for arachne-pnr is this:

> **Arachne-pnr is not maintained anymore; use [nextpnr](https://github.com/YosysHQ/nextpnr) instead, which is a complete functional replacement with major improvements.**

So the docs have already been updated...

closed 2023-10-29 by Ravenslofty